inthehand/32feet #142: BluetoothDevice.FromIdAsync  Why is it empty? I already know the ID of Bluetooth.

bbhxwl, 2021-07-26:
Why is it empty? I already know the ID of Bluetooth.

peterfoot, 2021-07-26:
What platform are you running on? For example on Windows the ID can be a device id which is different from a Bluetooth address. Please post the code you are using and some detail about the devices you are using.

bbhxwl, 2021-07-26:
> What platform are you running on? For example on Windows the ID can be a device id which is different from a Bluetooth address. Please post the code you are using and some detail about the devices you are using.

I have three Bluetooth devices, two of which can be connected by ID, but one can't, but this device can be paired manually, and then get the devices in the pairing list, and then connect. It's strange.

peterfoot, 2021-07-26:
Not all devices can be connected without pairing. The library doesn't provide programmatic pairing for Bluetooth LE devices. It's certainly something which is possible on most platforms and so I hope to add it in a future version.

bbhxwl, 2021-07-27:
> Not all devices can be connected without pairing. The library doesn't provide programmatic pairing for Bluetooth LE devices. It's certainly something which is possible on most platforms and so I hope to add it in a future version.

Sometimes, for example, after searching, you can use ID connection?