inthehand/32feet #143: How to get SCP Info

DatenThielt, 2021-07-26:
Afternoon, I am connecting to a controller and I am looking to browse the attributes for the InstalledServices,  However I can only seem to get the GUID , and cannot find a way to use that GUID to get the Attributes information I am looking for,

Did I miss something?
Thanks!

DatenThielt, 2021-07-27:
Just an update I can see this was available with the 3.5-3.7 Libaries but I would love to be able to do this in the latest version of the library, however sill have not found the GetServiceRecords(guid record) or its equivalent

bl1980, 2022-03-29:
I was also trying to use the GetServiceRecords method with the new libraries but realized it is not available and couldn't find an equivalent.  InstalledServices doesn't do the same thing.

DavidR5, 2023-07-05:
I have also been struggling with GetServiceRecords not being available in the current version. I am trying to identify if the remote server is listening on the ObexObjectPush service either using sdp or ServiceName. How can I achieve this?

peterfoot, 2023-07-06:
Hi, while it doesn't currently have the full service records available, you can get a list of services using BluetoothDeviceInfo.GetRfcommServicesAsync which will return a list of UUIDs of services supported - you can check this for ObexObjectPush to see if the server supports that service.