inthehand/32feet #145: A bug is found. Non null judgment does not work

bbhxwl, 2021-07-29:
A bug is found. Non null judgment does not work
Clearly empty, why would you enter this judgment?

[image]

bbhxwl, 2021-07-29:
[image]
Delegate of instance method cannot have empty 'this'

bbhxwl, 2021-07-29:
Delegate of instance method cannot have empty 'this'

bbhxwl, 2021-07-29:
Disconnect and reconnect. Do I still need to get characteristics again? Do you still need to reset characteristicvaluechanged? Do you still need to get the service again?
[image]

peterfoot, 2021-07-29:
Thanks, I've fixed the issue - there was an issue in the explicit operator for BluetoothDevice where when the native BluetoothLEDevice is null it returns an invalid object. It's checked into the code but not currently in a release.
Regarding the other issue, yes when reconnecting you should get the characteristics again.

bbhxwl, 2021-07-29:
> Thanks, I've fixed the issue - there was an issue in the explicit operator for BluetoothDevice where when the native BluetoothLEDevice is null it returns an invalid object. It's checked into the [...]

Hope you can update nuget, thank you. C # except for your project, there seems to be no better project across WinForm + xamarin.

bbhxwl, 2021-07-29:
> Hope you can update nuget, thank you. C # except for your project, there seems to be no better project across WinForm + xamarin.

There is also a strange bug. I have a Bluetooth device. If I disconnect and turn on the Bluetooth device again, it will enter the disconnection event. However, I don't need to reconnect. It will become connected in a moment. This is very strange. Is it a Bluetooth hardware problem? And when searching Bluetooth, can it be faster? Or asynchronous search? Because the search seems to take 30 seconds.

peterfoot, 2021-10-08:
Searching for new devices can be a slow process. If possible you can store the id of a known device and then retrieve this using BluetoothDevice.FromIdAsync(). Alternatively if you want to quickly get a list of paired devices you can use Bluetooth.GetPairedDevicesAsync().

bbhxwl, 2021-10-08:
> want

I am now familiar with the specific operation. I told the friend above about what I encountered, but it is really not as fast as the system to search for new devices. It takes about 30 seconds to search all new devices.