inthehand/32feet #148: DiscoverDevices issues Win 10

calgary123, 2021-08-05:
Hi,
I am quite sure this was working months ago....maybe Windows update broke it. I have no idea.

1.  Visual Studio 2019 Project targeting .NET 4.6.1, latest inthehand 4.0.20 according to nuget.  4.0.0.0 according to dll properties.

This code no longer finds anything discoverable and in fact only executes for 1-4 seconds (I expect DiscoverDevice to take 10 seconds by default?).  bdlist is always empty.

            var bc = new BluetoothClient();
            var bdlist = bc.DiscoverDevices();                           // this function only runs for a couple seconds, 0 to 4

2.  On the same computer I have my old project running visual studio 2017, targeting .NET 4.5, 32feet.NET version v3.5.0 according to nuget. 
This code returns discoverable devices.

            var bc = new BluetoothClient();
            var bdlist = bc.DiscoverDevicesInRange();                    // takes 10 seconds

3.  The two cases above are running on the same computer. I can switch between VS 2017  and 2019.  The first case never find any bluetooth devices, the second case always does.

4.  The visual studio 2019 project, I downgraded to the 32feet.net dll version 3.5.0.  Does not work.

Any idea what is going on here?

Appreciate your help.
Chris

peterfoot, 2021-08-05:
Have you changed the InquiryLength property at all? This will affect how long the discovery takes and if too short it may not pick up devices you are expecting to find. Also does the PairedDevices property return anything or is this also empty?

calgary123, 2021-08-05:
Hi Peter,
Good news, adding debug output led me to follow some exceptions and figure out what is going on.  I have found the cause, although I am not entirely sure what to do about it.

This software is used for testing product, The new version (Visual Studio 2019) is a lot more asynchronous (async, await, etc).  It is attempting to open serial ports for communication in another task at the same time as the Bluetooth scan starts in another thread. 

This hasn't been a problem but I notice that two new serial ports have appeared in the device manager called 'Standard Serial over Bluetooth link (COM12)' and COM13.  Attempting to open these results in System.IO exceptions. No big deal one would think.

But if these happen (in another task) after calling DiscoverDevices (in its own task) this seems to cause DiscoverDevices to abort/return before finishing.

So at this point I can find a way to work around whatever is happening beneath the hood. Unfortunately calling SerialPort.GetPortNames only returns COM12, COM13, etc... so I am not entirely sure yet how I will figure out which ones are 'bluetooth' and skip them.

Thanks for your quick reply.

Chris

peterfoot, 2021-08-05:
I'm not sure why the COM ports are affecting discovery - I'll have to try and recreate this and dig into it. You may find this post and the accompanying Gist useful which shows how to get details of Bluetooth virtual COM ports 

https://inthehand.com/2020/11/22/bluetooth-virtual-com-ports/