inthehand/32feet #152: A socket operation failed because of the destination host was down

Echarnus, 2021-09-04:
I'm getting the following error

`System.Net.Internals.SocketExceptionFactory.ExtendedSocketException: 'A socket operation failed because the destination host was down. 38184C066AAF:0000110b-0000-1000-8000-00805f9b34fb (0)'`

Which honestly doesn't give a lot of information, especially since the _DeviceInfo_  shows it's connected and I'm using a service installed in the bluetooth installed services.

This is my test code.

```
var cli = new BluetoothClient();
var bluetoothDeviceInfos = cli.PairedDevices;
var deviceInfo = bluetoothDeviceInfos.FirstOrDefault(_ => _.DeviceName.Contains("WH-1000XM3"));
cli.Connect(deviceInfo.DeviceAddress, deviceInfo.InstalledServices.ElementAt(0));
```

Any info on what's going wrong? All I want to do is open a stream and read the data sent by the device.

peterfoot, 2021-09-04:
It's difficult to tell what the issue might be but a few things to investigate:-
InstalledServices is cached so although these may be listed they may not be currently enabled on the remote device, indeed the device could be offline and will still show in PairedDevices and have a cached list of InstalledServices.
Another possibility is that there is an open connection on the target service to another device or another connection which was not successfully closed. Resetting the devices or toggling Bluetooth off/on may clear this.

Echarnus, 2021-09-05:
I was at the moment listening to some music with my headset, on the same device I was running the code on. Wouldn't it be possible to have both running my app and using the headset at the same time? If not, how does Wireshark sniff the data then while still being able to use the headset? 

Have tried all of the InstalledServices though.

Echarnus, 2021-09-15:
Just bumping this thread because I'm honestly stilling running into the same problem.