inthehand/32feet #153: BluetoothGattCharacteristic have only 20 byte (default) for read

polipo, 2021-09-10:
Android for default have BluetoothGattCharacteristic set to 20 byte,
see https://stackoverflow.com/questions/24135682/android-sending-data-20-bytes-by-ble

For avoid this will be set the BluetoothGattService.RequestMtu(MYSIZE)

Question : 
How to set by InTheHand.BluetoothLE ?

Thanks

peterfoot, 2021-09-11:
Currently this would only be possible in platform specific code - cast the GattService to Android.Bluetooth.BluetoothGattService and call RequestMtu. Looking at adding a method to get/set the MTU into the cross-platform API.

polipo, 2021-11-09:
The Android have some problems to set a different value, it require to set twice my personal workaround is to set X-1 and then set X.

peterfoot, 2021-11-15:
I found this in the WebBluetooth issues which details observations from different platforms - https://github.com/WebBluetoothCG/web-bluetooth/issues/284

peterfoot, 2021-12-21:
In the latest release I've added a property to query the MTU value, also on Android it requests 512bytes and will negotiate a value with the remote device so it is more in the range of what you see with other platforms

closed 2021-12-21 by peterfoot