inthehand/32feet #157: Xamarin.iOS Bluetooth to W10 PC Support?

danielricci, 2021-10-20:
Hi.

I have a Xamarin app that has both an Android and iOS project (code is written in C#), and I'm able to use your library to communicate with my other C# application (through bluetooth) on my W10 PC when running your code on Android, but not on iOS. For example when I make a call to BluetoothClient.PairedDevices it returns values when running on Android, but on iOS its always empty.

Maybe i'm doing something wrong with the library, or maybe it's not supported?

peterfoot, 2021-10-20:
The iOS part of the library is not fully implemented yet. iOS doesn't have a full API for Bluetooth Classic and so it uses the ExternalAccessory API which only supports certain device types which are MFI certified by Apple.

danielricci, 2021-11-01:
Can I accomplish this using Bluetooth LE portion of your library?

peterfoot, 2021-11-09:
Bluetooth LE is supported on iOS yes, but only the client side (consuming services/characteristics on another device) and not currently the server side (publishing services/characteristics).

bbhxwl, 2022-03-10:
> Bluetooth LE is supported on iOS yes, but only the client side (consuming services/characteristics on another device) and not currently the server side (publishing services/characteristics).

I don't know why my IOS can't connect successfully. Can you give me some advice?

danielricci, 2022-03-10:
Closing issue.

closed 2022-03-10 by danielricci