inthehand/32feet #159: Android Gatt AutoConnect is always set to true

polipo, 2021-11-09:
The PlatformInit for Android, force the connection to autoconnect = true, this parameter is not always correct,
in my case I will control the disconnection and reconnection.

 private void PlatformInit()
        {
            _gattCallback = new GattCallback(this);
           _gatt = ((ABluetooth.BluetoothDevice)Device).ConnectGatt(Android.App.Application.Context, **true**, _gattCallback, ABluetooth.BluetoothTransports.Le);
        }

peterfoot, 2021-11-09:
Thanks, good catch. Going forward I think the required behaviour is to set this false and allow the consumer of the library to manage reconnection, using events to detect a disconnection.

polipo, 2021-11-09:
I have pulled a request for a change.
Thanks

peterfoot, 2021-11-15:
Pull request #160 merged.

closed 2021-11-15 by peterfoot