inthehand/32feet #164: Pairing a device in Unity freezes the editor afterwards

ExcellentBrain, 2021-12-01:
I recently discovered 32Feet and I love how easy to use it is. I use a Bluetooth device as a source of input for my Unity project, but I noticed an issue that pops up after my code runs BluetoothSecurity.PairRequest.

After running a pair request and exiting play mode, I noticed that any action that requires me to refresh Unity causes it to get stuck. For example, if I try to enter play mode again, it gets stuck on Application.Reload or if I need to recompile my project after updating a script, it gets stuck on Reloading Script Assemblies.

I tried running it both synchronously (just directly calling PairRequest as well as asynchronously using a Task.Run, but both produce the same issue described.

```
private async Task<bool> AttemptPairAsync(BluetoothDeviceInfo btDevice, string pin = null)
{
    Func<bool> func = () =>
    {
        if (btDevice.Authenticated)
           return true;
        CheckCancellation();
        bool result = BluetoothSecurity.PairRequest(btDevice.DeviceAddress, pin);
        btDevice.Refresh();
        return result;
    };
    return await Task.Run(func);
}
```

My Unity version is 2020.3.19f1
My InTheHand.Net.Bluetooth package is version 4.0.22 and was added using NuGet for Unity.

peterfoot, 2021-12-01:
This may be related to #162 and sounds like a problem with the callback function. Currently investigating...

ExcellentBrain, 2021-12-01:
Alright, thank you for the fast response. I hope you find the solution soon.

peterfoot, 2021-12-01:
There was an issue in the callback code. Fixed in 4.0.24