inthehand/32feet #166: InTheHand Bluetooth and .NET 5 and 6

polipo, 2021-12-14:
Compatibility with frameworks;
The nuget package BluetoothLE seem to be compatible with .NET 5, it can be installed on .NET 5/6 class library
but when i compile it shows this error

Severity	Code	Description	Project	File	Line	Suppression State
Error	NETSDK1149	InTheHand.BluetoothLE.dll cannot be referenced because it uses built-in support for WinRT, which is no longer supported in .NET 5 and higher.  An updated version of the component supporting .NET 5 is needed. For more information, see https://aka.ms/netsdk1149	WpfApp17	C:\Program Files\dotnet\sdk\6.0.100\Sdks\Microsoft.NET.Sdk\targets\Microsoft.NET.Sdk.targets	1003	

Can be in next releases (if possible) the compatibility with .NET 5/6 ?

peterfoot, 2021-12-14:
The current release I think includes a .NET Core 3.1 build and not 5.0 and it is using this but it is not compatible. I've been testing a .NET 5.0 build which uses the new approach and should therefore support both 5.0 and 6.0. I'll build and release an update shortly.

peterfoot, 2021-12-21:
This should now be resolved, please try 4.0.22

polipo, 2021-12-23:
Dear Peter, I have tested the 4.0.22 
My environment is VS2022Enterprise (last update at present date), I have build a ClassLibrary .net 5.0 and installed the BlueToothLe nuget package 4.0.22. It succesfully install on it, during build the VS says:

Severity	Code	Description	Project	File	Line	Suppression State
Error	NETSDK1149	InTheHand.BluetoothLE.dll cannot be referenced because it uses built-in support for WinRT, which is no longer supported in .NET 5 and higher.  An updated version of the component supporting .NET 5 is needed. For more information, see https://aka.ms/netsdk1149	LibNet5	C:\Program Files\dotnet\sdk\6.0.101\Sdks\Microsoft.NET.Sdk\targets\Microsoft.NET.Sdk.targets	1003

peterfoot, 2021-12-23:
I think the reason it is picking the wrong dll is possibly because of the target sdk and minimum sdk defined for InTheHand.BluetoothLE versus what the consuming project uses. For 4.0.22 it only produces a .NET 5.0 dll targetting net5.0-windows10.0.19041.0 with the minimum platform set to 10.0.18362.0. I need to recheck if this is a valid cut off (there are some API changes in various Windows 10 versions). If your consuming project supports an older version it may be selecting the .NET Core dll incorrectly.

polipo, 2021-12-23:
Yes exactly, I have removed my last comment because I found same as you tell. Now I'm try some test about it. Thanks

yetyman, 2021-12-28:
Just a note for people like me that arrived here and did not know the solution. you can hardcode the windows version into your csproj's TargetFramework like so to force it to choose the correct platform version and escape your winRT woes
`
    <TargetFramework>net5.0-windows10.0.19041.0</TargetFramework>
    `

peterfoot, 2022-03-27:
I've added a .NET 6.0 Windows build in 4.0.23 too.

closed 2022-03-27 by peterfoot

FifthCloud, 2022-10-28:
Hey Peter,
Trying to figure out what was the answer here. Here is what I have in my csproj and let me know what I am missing because I am getting the error message discussed.
`<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <TargetFramework>net6.0</TargetFramework>
    <ImplicitUsings>enable</ImplicitUsings>
    <Nullable>enable</Nullable>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference Include="InTheHand.BluetoothLE" Version="4.0.25" />
  </ItemGroup>

</Project>`

Should I specify my windows builds?

Thank you.

peterfoot, 2022-10-31:
There isn't a generic .NET 6.0 dll - so your consuming project has to reference a platform target framework to get the correct dll. If the project is just net6.0 it will fall back to an older framework version and you'll get errors about the interop with Windows APIs. See the reply from @yetyman above for the workaround.