inthehand/32feet #171: SetPin is missing

peterfoot, 2022-01-31:
Migrated from Uservoice

"I have upgraded to version 4.0.25 and I am missing the SetPin method on the BluetoothClient.
I have for years been using a very old version of the software and will now upgrade everything to .net 6. Therefor I have installed version 4.0.25.

The smart thing about the old version was that I could apply the Pin Code with SetPin and it was used when needed, with no interaction of the user.

Could you please give me any hit on how to use the new implementation in this regard.

I must say that I am very happy for and impressed with your Bluetooth implementation, because it does not mess around with virtual COM ports and MicroSoft does."

peterfoot, 2022-01-31:
SetPin didn't really work across platforms so the BluetoothSecurity.PairDevice provides a standard pairing API across the different implementations. However it could possibly be added to the Win32 only implementation...

ErikBis, 2022-01-31:
Thank you, but the method made my life so much easier :-) 
Now I have to check for pairing before connecting I guess
Thank you for your fast reply