inthehand/32feet #172: BeginConnect throws an exception

ErikBis, 2022-02-01:
I get an Exception when I try to connect to Bluttoth
Following are giving me an exception ()
            var addr = BluetoothAddress.Parse("00025BAABB99");
            var service = BluetoothService.SerialPort;
            var port = 2;

            var ePoint = new BluetoothEndPoint(addr, service, port);
            var client = new BluetoothClient();
            client.BeginConnect(ePoint, ar => { }, new object());

I have upgraded to version 4.0.25 and now I get this Exception
I not sure what I do wrong,

ErikBis, 2022-02-01:
I have the same problem with ConnectAsync

peterfoot, 2022-02-01:
What is the exception? It should contain the native SocketException which will help identify the reason

ErikBis, 2022-02-01:
This is what I get:
System.NullReferenceException
  HResult=0x80004003
  Message=Object reference not set to an instance of an object.
  Source=InTheHand.Net.Bluetooth
  StackTrace:
   at InTheHand.Net.Sockets.BluetoothClient.BeginConnect(BluetoothEndPoint endpoint, AsyncCallback requestCallback, Object state)
   at WireOnAir.Infrastructure.Connections.Bluetooth.ClientWrapper.BtClientWrapper.<BeginConnect>d__6.MoveNext() in C:\Users\ebs\Source\Repos\WoaEnvironmentBuilder\Infrastructure\Connections\Bluetooth\ClientWrapper\BtClientWrapper.cs:line 48

ErikBis, 2022-02-01:
I can not see the native SocketException

peterfoot, 2022-02-01:
Okay, I can see the issue, the client socket is not created in the BeginConnect method, but is in the sync Connect method. I'll get this resolved for the next release.

ErikBis, 2022-02-01:
Thank you for your fast reply

peterfoot, 2022-02-02:
Fixed in 4.0.26 now up on NuGet

closed 2022-02-02 by peterfoot

ErikBis, 2022-02-06:
I have now tested the BeginConnect and I don't get an exception, but I don't get connected either
When I call EndConnect with the IAsyncResult I get the following exception
As an remark I get an connection when I use Connect (_client.Connect(_deviceAddress, BluetoothService.SerialPort);)

System.Net.Sockets.SocketException
  HResult=0x80004005
  Message=An invalid argument was supplied.
  Source=System.Private.CoreLib
  StackTrace:
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Threading.Tasks.TaskToApm.End(IAsyncResult asyncResult)
   at System.Net.Sockets.Socket.EndConnect(IAsyncResult asyncResult)
   at InTheHand.Net.Sockets.BluetoothClient.EndConnect(IAsyncResult asyncResult)
   at WireOnAir.Infrastructure.Connections.Bluetooth.ClientWrapper.BtClientWrapper.EndConnect(IAsyncResult asyncResult) in C:\Users\ebs\Source\Repos\WoaEnvironmentBuilder\Infrastructure\Connections\Bluetooth\ClientWrapper\BtClientWrapper.cs:line 60
   at WireOnAir.Infrastructure.Connections.Common.Clients.ClientAsync.ConnectedCallback(IAsyncResult result) in C:\Users\ebs\Source\Repos\WoaEnvironmentBuilder\Infrastructure\Connections\Common\Clients\ClientAsync.cs:line 166