inthehand/32feet #175: Set PHY trasmission on BluetoothLE for iOS

polipo, 2022-03-02:
In the BluetoothLE for iOS miss the implementation of  "PreferredPhy" properties on Gatt implementation on InTheHand library.

I have found, on above link, an example how to set the preferred phy, it seems strange this 
will apply on BTLe's Characteristics and not into the Gatt.

What do you think about this and can be a possible future implementation?
 https://github.com/NordicPlayground/iOS-Image-Transfer-Demo/blob/master/imageTransfer/Bluetooth/CameraPeripheral.swift

 if (Is2MPhy == true)
            {
                byte[] data = new byte[2] { 0x05, 0x02 }; // 0x05=Physic; 0x02 = 2M Phy
                _commandCharacteristicRead.WriteValueWithResponseAsync(data);
                _commandCharacteristicWrite.WriteValueWithResponseAsync(data);
                _realtimeCharacteristicRead.WriteValueWithoutResponseAsync(data);
                //https://github.com/NordicPlayground/iOS-Image-Transfer-Demo/blob/master/imageTransfer/Bluetooth/CameraPeripheral.swift
                _deviceLE.Gatt.PreferredPhy = InTheHand.Bluetooth.BluetoothPhy.Le2m; //isnt implemeted
                Debug.WriteLine("_deviceLE.Gatt.PreferredPhy=2Mbit");
            }

peterfoot, 2022-03-02:
That appears to be a Nordic specific implementation and is not a way to set the Phy on any remote device. Unfortunately there is no mechanism in CoreBluetooth to set the Phy or retrieve the current Phy being used, even though iOS supports Bluetooth 5.

polipo, 2022-03-09:
I have done some tests about iOS and the bluetotth LE connections:
iPad  9'generation (bluetooth 4.2)
iPad  mini 6'generation (bluetooth 5.0)

The iPad 9'gen having bluetooth 4.2 doesn't support the LE Phy 2Megabit and otherwise the iPad mini 6'gen. having the
bluetooth 5.0 supports automatically the Phy 2Megabit.

I confirm the iOS dialogue successfully in Phy 2M with a device which supports it, automatically set if the iOS have BT >= 5.0

bbhxwl, 2022-03-10:
> In the BluetoothLE for iOS miss the implementation of "PreferredPhy" properties on Gatt implementation on InTheHand library.
>
> I have found, on above link, an example how to set the preferred phy, it [...]

I can't use xamarin Forms. Do you know why?

peterfoot, 2022-03-10:
@polipo yes the devices should auto-negotiate when both support the phy type. A peripheral can express a preferred phy and that will be used if possible. On Android you are able to request a phy at the central (Android device) side and this can override the setting. I'm going to close this issue for now - I'll certainly revisit the functionality if CoreBluetooth changes.

closed 2022-03-10 by peterfoot