inthehand/32feet #177: IOS can't use Bluetooth either. I only tested it successfully in win10 system. I don't know the reason.

bbhxwl, 2022-03-10:
bool availability = await Bluetooth.GetAvailabilityAsync();
The above code has always been false. I added the following parameters. Bluetooth is also on. Do not know why?

<key>NSBluetoothPeripheralUsageDescription</key>
	<string>This app requires Bluetooth to demonstrate the Bluetooth LE APIs</string>
	<key>NSBluetoothAlwaysUsageDescription</key>
	<string>This app requires Bluetooth to demonstrate the Bluetooth LE APIs</string>

peterfoot, 2022-03-10:
Have you added the NuGet package to the iOS project? What are you deploying to (it must be a physical device)?

bbhxwl, 2022-03-10:
> Have you added the NuGet package to the iOS project? What are you deploying to (it must be a physical device)?

I am a physical device, iPad 2021 and iPhone 12. I want to realize Android + IOS cross platform Bluetooth communication. Can the author realize the method of asynchronous search for Bluetooth devices?

I can't use Android and IOS. I don't know why. But I can search by using [Install-Package Plugin.BLE] IOS. Android can't.
Can you help me?