inthehand/32feet #178: Could not load signature of InTheHand.Net.Bluetooth.Droid.AndroidNetworkStream:WriteAsync

gohan-oishii, 2022-03-20:
Hi,
I'm using inthehand in .NET MAUI application to send data to Android device via Bluetooth, like this.

```
using BluetoothClient client = new BluetoothClient();
client.Connect(new BluetoothAddress(bd_addr), BluetoothService.SerialPort);
NetworkStream bt_stream = client.GetStream();
bt_stream.Write(data, 0, data.Length);
bt_stream.Close();
```

However, I faced an exception that is "Could not load signature of InTheHand.Net.Bluetooth.Droid.AndroidNetworkStream:WriteAsync" from GetStream method.
Since inthehand has been used for Xamarin and Android, I thought .NET MAUI is also supported.
Could you tell how to resolve this?

peterfoot, 2022-03-21:
I think this may be because there is currently a monoandroid build of the library and a .NET 6 windows version but not a .NET 6 Android build. I will need to add this to the NuGet package.

peterfoot, 2022-09-11:
This is resolved now with latest NuGet package.

closed 2022-09-11 by peterfoot