inthehand/32feet #183: Set the NamePrefix option

Yanis131, 2022-03-28:
Hey,

I'm doing a scan for BLE devices with setting a NamePrefix filter but it isn't working.

At this line https://github.com/inthehand/32feet/blob/9517bd5bcc4206e76b892a43d6c6f5ba1dff36d2/InTheHand.BluetoothLE/Platforms/Windows/Bluetooth.windows.cs#L124

We can clearly see that the AQS is a plain text string with no references to the `option` object.

Shouldn't the AQS string include the `NamePrefix` provided in the option object ?

If so, I would be down to make a PR 😃 

---

<details>
  <summary>My code</summary>

```cs
using InTheHand.Bluetooth;
using System;
using System.Collections.Generic;
using System.Diagnostics;
using System.Threading.Tasks;

namespace _32feet_Demo
{
    public class Program
    {
        public static async Task Main(string[] args)
        {
            var opts = new RequestDeviceOptions();
            
            opts.Filters.Add(new BluetoothLEScanFilter
            {
                NamePrefix = "PREFIX-" 
            });

            var stopWatch = new Stopwatch();
            stopWatch.Start();

            var devices = await Bluetooth.ScanForDevicesAsync(opts);

            foreach (var device in devices)
            {
                Console.WriteLine(!string.IsNullOrEmpty(device.Name) ? device.Name : "???");
            }

            Console.WriteLine($"Done in {stopWatch.Elapsed.TotalSeconds} seconds");
        }
    }
}
```
  
</details>

---

Thank you
-Yanis Bouchemot

peterfoot, 2022-04-23:
I'm not sure whether we can pass this in the AQS string, it would need to be implemented on top of this and filter the returned device collection.

Yanis131, 2022-04-27:
Hey,

It is actually possible natively with no extra parsing required with [query operators ](https://docs.microsoft.com/en-us/windows/win32/search/-search-3x-advancedquerysyntax#query-operators)using the `COP_VALUE_STARTSWITH` operator.

PR #188 (2022-04-28): Add the NamePrefix filter in the AQS string
Fixes #183
closed 2022-04-28 by peterfoot