inthehand/32feet #184: I got PlatformNotSupportedException when using InTheHand.BluetoothLE in Unity

hieuvu2719, 2022-03-29:
Hi!
I built my project with Unity editor version 2020. I have successfully imported InTheHand.BluetoothLE.dll version 4.0.22 into that same project. However, when I used Bluetooth.ScanForDeviceAsync, I got a PlatformNotSupportedException, which said that this operation is not supported on this platform. InTheHand.BluetoothLE.dll targets .NET 4.x and my Unity runtime also supports .NET 4.x so I'm not sure what went wrong. I'd really appriciate if you can help me. Thank you in advance!

peterfoot, 2022-03-31:
How are you adding the dll to your Unity project? Have you tried extracting the correct dll manually from the nupkg and referencing that?

igorbasko01, 2023-03-02:
Hey, I've encountered a similar issue.
When I install the package using some nuget package manager for unity, it copies the netstandard dll.
When I run the unity project and try to do the following:
```csharp
using InTheHand.Bluetooth;
using UnityEngine;

public class BLEController : MonoBehaviour
{
    // Start is called before the first frame update
    async void Start()
    {
        var devices = await Bluetooth.ScanForDevicesAsync();
        Debug.Log(devices);
    }

    // Update is called once per frame
    void Update()
    {
        
    }
}
```
I get the `PlatformNotSupportedException: Operation is not supported on this platform.` exception.
If I manually replace the `netstandard` with `net461`, I get the following errors in the Unity Editor:
```
Assembly 'Assets/Packages/ble/net461/InTheHand.BluetoothLE.dll' will not be loaded due to errors:
Unable to resolve reference 'Windows.Foundation.UniversalApiContract'. Is the assembly missing or incompatible with the current platform?
Reference validation can be disabled in the Plugin Inspector.
Unable to resolve reference 'System.Runtime.WindowsRuntime'. Is the assembly missing or incompatible with the current platform?
Reference validation can be disabled in the Plugin Inspector.
Unable to resolve reference 'Windows.Foundation.FoundationContract'. Is the assembly missing or incompatible with the current platform?
Reference validation can be disabled in the Plugin Inspector.
```
I tried changing the player settings to use `.NET Framework`, which didn't help.

I'm using Unity on Windows 11 Pro.

I'm not sure what else I can do. Would appreciate some help.

peterfoot, 2023-03-02:
I'm not sure if NuGet in Unity pulls in all the dependencies in the same way as Visual Studio but can you manually add the Microsoft.Windows.SDK.Contracts package as this is required for the .NET Framework dll.

igorbasko01, 2023-03-02:
Hey, thanks for the suggestion. 
It seems that when I change the Api Compatibility Level to .NET Framework, the [nuget package manager](https://github.com/GlitchEnzo/NuGetForUnity) is installing the correct dependencies.
[image]
[image]
Thing is, it looks like not the whole dependency is installed. But it seems that it copies what it thinks is relevant.
For example the `System.Runtime.WindowsRuntime.4.6.0` the `lib\net45` folder is actually empty also in the full package as well.
[image]
[image]
I copied the whole `Microsoft.Windows.SDK.Contracts` into unity:
[image]
And the error did disappear, but now when I run the [above code](https://github.com/inthehand/32feet/issues/184#issuecomment-1451454778), it fails with: `TypeLoadException: Failure has occurred while loading a type.`
I tried adding also dlls for netstandard but still getting the TypeLoadException when running the code.

Were you ever able to use BluetoothLE in Unity?