inthehand/32feet #187: PlatformNotSupportedException and System.DllNotFoundException: ws2_32.dll in 4.0.29

Adrien-pierret, 2022-04-26:
Hello everyone,

While trying to build and/or run the projects I ran into 2 distinct issues:
・If I run the project using .NET Standard 2.0 I get a "PlatformNotSupportedException"
・If I change and run it with .NET 4.x I get a "System.DllNotFoundException: ws2_32.dll" and a "System.Windows.Forms.dll assembly is referenced by user code, but is not supported on Android platform. Various failures might follow." warning.

Both errors occur for the "InTheHand.Net.Bluetooth 4.0.29" package when i try to execute:
`BluetoothClient client = new BluetoothClient();`
And they both happen on the phone and on the simulation.

Other info:
・I am running Unity 2020.3.33f1
・OS: Ubuntu 20.04
・Target: Android (Google pixel 4)
・The package was installed using Nuget for Unity

Best regards

peterfoot, 2022-04-26:
I think the way that Unity manages NuGet platforms is different to Visual Studio which leads it to select the wrong dll. There is a monoandroid dll in the package which should work. You could try extracting it manually and referencing it directly. The .NET 4 dll is for Windows only hence the ws2_32.dll error, the .NET Standard dll has no implementation and simply acts as an API surface to be replaced with the correct runtime version. I will need to do some more testing with Unity to work out a fix.

Adrien-pierret, 2022-04-26:
Thank you peter for such a quick answer.
So I gave it a try  and downloaded the package from "https://www.nuget.org/packages/InTheHand.Net.Bluetooth"
I just took the dll from the monoandroid11 folder and added it to my plugins and Assets/Packages (not at the same time, on 2 different occasions). Now when I try to build I run into the following:

"ArgumentException: The Assembly Mono.Android is referenced by InTheHand.Net.Bluetooth ('Assets/Plugins/inthehand.net.bluetooth.4.0.29/InTheHand.Net.Bluetooth.dll'). But the dll is not allowed to be included or could not be found."

I tried to compile for Android 10; 11 and "highest available" in Unity. I Also tried both .NET 2.0 and 4.X. Same result.
Am I missing something?

p.s: I am new to Unity and C# so I might have done something wrong. Sorry if I made some newbies mistakes.

peterfoot, 2022-04-27:
I'm not a native to Unity either so don't know my way around all the differences with a regular .NET project. I've used it previously to test the Windows version of the dll, but not Android but it _should_ work. I'll do some investigation...

NicholasLimHongDa, 2024-02-02:
I have the same problem, here's the environment I used,
・Unity 2022.3.16f1
・Dev OS: Windows 11 Home
・Target: Android (Google Pixel 7a)
・Package was installed using NuGet for Unity

And this is what I tried,
・.NET Standard on Unity, with netstandard2.0 dlls = PlatformNotSupported exception
・.NET Framework on Unity, with netstandard2.0 dlls = PlatformNotSupported exception
・.NET Standard on Unity, with netstandard2.1 dlls = PlatformNotSupported exception
・.NET Framework on Unity, with netstandard2.1 dlls = PlatformNotSupported exception
・.NET Standard on Unity, with net461 dlls = PlatformNotSupported exception
・.NET Framework on Unity, with net461 dlls = System.DllNotFoundException: ws2_32.dll
・.NET Standard on Unity, with monoandroid10.0 dlls = ArgumentException: The Assembly Mono.Android is referenced by InTheHand.Net.Bluetooth
・.NET Framework on Unity, with monoandroid10.0 dlls = ArgumentException: The Assembly Mono.Android is referenced by InTheHand.Net.Bluetooth

The closest I could get this to work was on ".NET Framework on Unity, with net461 dlls", but I couldn't get it to run on Android device, cause the package is still trying to reference dlls made for Windows. Is there a workaround or a possible solution for this?