inthehand/32feet #192: Use the component to obtain the battery information of the device corresponding to bluetooth, please help, thank you

jzeb, 2022-05-24:
I want to use the service to obtain the electric quantity information corresponding to bluetooth device, using the headset of the latest version 5.2 of Bluetooth. Under the platform of WIN11 operating system, the electric quantity information can be displayed normally, but I use the debugging software "nRF Connect" of mobile phone, and the screenshot is as follows. How do I obtain the electric quantity information of Bluetooth device now? Please give me an example. Thank you.
**
[image]
**

peterfoot, 2022-05-24:
If it supports the Battery service it will have a service with id 0x180F. The characteristic 0x2A19 contains a single byte with the battery percentage,

jzeb, 2022-05-25:
However, some (such as the part of my screenshot) do not have this service, but the bluetooth list of WIN11 can display the battery information, may I ask, is there any other way?