inthehand/32feet #193: A2DP and Hands-Free

AndreGroeschel, 2022-06-02:
Hi there,

I have a theoretical question...is it possible with your library to connect to a phone and make phone calls and stream music? 
Do we simply need to connect to the service and the rest is supported by Windows or do we have to do the low-level work and open an audio channel?

Appreciate any advice here, since we would like to implement this but right now we have no idea about the complexity behind this

Cheers, André

peterfoot, 2022-06-06:
For Windows 10 and above there are WinRT APIs for both hands free calling and audio. I've used these in my Thy Phone app but there isn't currently a wrapper API in 32feet.NET.

AndreGroeschel, 2022-06-07:
Hi Peter,

thank you very much for the info!!

Are you referring to these APIs [](https://docs.microsoft.com/en-us/uwp/api/windows.applicationmodel.calls?view=winrt-22621) ?

peterfoot, 2022-06-07:
Yes, that allows you to use a handsfree paired phone.