inthehand/32feet #194: BluetoothRadio.Default returning null.

wagenheimer, 2022-06-02:
Hi!

I need to connect to a Bluetooth Device and receive ASCII data from it. Something very simple.

I'm trying a simple sample, but it fails because  var radio = BluetoothRadio.Default; is returning null.

I have all permissions 
[assembly: UsesPermission(Android.Manifest.Permission.BluetoothScan)]
[assembly: UsesPermission(Android.Manifest.Permission.Bluetooth)]
[assembly: UsesPermission(Android.Manifest.Permission.BluetoothAdmin)]
[assembly: UsesPermission(Android.Manifest.Permission.BluetoothConnect)]
[assembly: UsesPermission(Android.Manifest.Permission.AccessCoarseLocation)]

I couldn't find anything in the documentation about what could be wrong. Could you help me to find out what is missing?

wagenheimer, 2022-06-02:
BluetoothClient client = new BluetoothClient(); also raised an exception.

```
Not Flagged	>	1	0	Worker Thread		Android.Runtime.JNINativeWrapper._unhandled_exception(Java.Lang.SecurityException e) Line 12
 	 	 	 	 	 	0xE in Android.Runtime.JNINativeWrapper._unhandled_exception at /Users/runner/work/1/s/xamarin-android/src/Mono.Android/Android.Runtime/JNINativeWrapper.g.cs:12,5
 	 	 	 	 	 	0x1D in Android.Runtime.JNINativeWrapper.Wrap_JniMarshal_PP_V at /Users/runner/work/1/s/xamarin-android/src/Mono.Android/Android.Runtime/JNINativeWrapper.g.cs:23,26
 	 	 	 	 	 	0x12 in System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw at /Users/builder/jenkins/workspace/archive-mono/2020-02/android/release/mcs/class/referencesource/mscorlib/system/runtime/exceptionservices/exceptionservicescommon.cs:157,13
 	 	 	 	 	 	0x6 in System.Runtime.CompilerServices.AsyncMethodBuilderCore.<>c.<ThrowAsync>b__7_0 at /Users/builder/jenkins/workspace/archive-mono/2020-02/android/release/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/AsyncMethodBuilder.cs:1021,49
 	 	 	 	 	 	0xC in Android.App.SyncContext. at /Users/runner/work/1/s/xamarin-android/src/Mono.Android/Android.App/SyncContext.cs:36,19
 	 	 	 	 	 	0xE in Java.Lang.Thread.RunnableImplementor.Run at /Users/runner/work/1/s/xamarin-android/src/Mono.Android/Java.Lang/Thread.cs:36,6
 	 	 	 	 	 	0x8 in Java.Lang.IRunnableInvoker.n_Run at /Users/runner/work/1/s/xamarin-android/src/Mono.Android/obj/Release/monoandroid10/android-31/mcw/Java.Lang.IRunnable.cs:84,4
 	 	 	 	 	 	0x8 in Android.Runtime.JNINativeWrapper.Wrap_JniMarshal_PP_V at /Users/runner/work/1/s/xamarin-android/src/Mono.Android/Android.Runtime/JNINativeWrapper.g.cs:22,5
```

wagenheimer, 2022-06-02:
I this this is related to - #https://github.com/inthehand/32feet/issues/185

peterfoot, 2022-10-10:
Depending on your target Android version you need to request the Bluetooth permissions at runtime as well as in the manifest. For .NET MAUI there is the start of the library for this here - https://github.com/inthehand/32feet/tree/main/InTheHand.Bluetooth.Permissions
Eventually this will be released as a standalone NuGet package.

Speedberg, 2023-08-09:
I had the same issue and fixed it by doing the following in my main activity:

```
protected override void OnCreate(Bundle savedInstanceState)
{
   //Make sure to set this
   InTheHand.AndroidActivity.CurrentActivity = this;

   base.OnCreate(savedInstanceState);
}
```

For more info: https://github.com/inthehand/android-activity