inthehand/32feet #195: Accept incoming call from Bluetooth Headset on Windows

pqviet07, 2022-06-06:
Hi all, currently I am developing a feature (accept/reject call from bluetooth headset) for a softphone on Windows, can I use this library to capture events from Bluetooth Headset (airpod, galaxy buds, jabra elite, ...) like press buttons such as accept/reject call?

peterfoot, 2022-06-06:
Are you using the hands free device for audio? If so when paired and connected Windows will be using the control service itself as well as the dedicated audio channel. When you press the action buttons I think these are converted into specific keyboard events but I haven't looked into this yet.

pqviet07, 2022-06-06:
> Are you using the hands free device for audio? 

Yes, I also tried create "Service Level Connection Initialization Procedure" that described in Hands-free Profile Specification
[link](https://www.bluetooth.com/specifications/specs/hands-free-profile-1-8/). But I just recv first command: ``AT+BRSF=923``

[image]

this is my code:
```C#

            BluetoothDeviceInfo device = new BluetoothDeviceInfo(0xDCCCE660788F); // hardcode for MAC address
            BluetoothClient client = new BluetoothClient();

            device.SetServiceState(BluetoothService.Handsfree, false);
            client.Connect(device.DeviceAddress, BluetoothService.Handsfree);

            if (client.Connected)
                MessageBox.Show("OK");

            NetworkStream stream;

            stream = client.GetStream();
            if (stream.CanRead)
            {
                byte[] buff = new byte[1024];
                int n = 0;
                StringBuilder str = new StringBuilder();
                while (stream.DataAvailable)
                {
                    n = stream.Read(buff, 0, buff.Length);
                    str.AppendFormat("{0}", Encoding.ASCII.GetString(buff, 0, n));
                }
                MessageBox.Show(str.ToString()); // I recv: AT+BRSF=923 

                // send ACK
                if(stream.CanWrite)
                {
                    string temp1 = "+BRSF=923\r\n";
                    buff = Encoding.ASCII.GetBytes(temp1);
                    stream.Write(buff, 0, temp1.Length);

                    string temp2 = "OK\r\n";
                    buff = Encoding.ASCII.GetBytes(temp2);
                    stream.Write(buff, 0, temp2.Length);
                }

                while (stream.DataAvailable)
                {
                    n = stream.Read(buff, 0, buff.Length);
                    str.AppendFormat("{0}", Encoding.ASCII.GetString(buff, 0, n));
                }
                MessageBox.Show(str.ToString()); // continue recv: AT+BRSF=923 (this is wrong with flow in spec)
            }

```

peterfoot, 2022-06-06:
Yes, if the OS is using the connection it would be better not to create another one. I'm looking into this at the moment as I have some code to intercept the media controls (Play/Pause Next/Previous) on a headset so just looking to see if this can be used for call control too. It's possible to intercept these as keyboard events on Win32 and WinUI projects.

jzeb, 2022-06-08:
Yes, I got the battery information of the earphone and used the code you said (WINDOWS needs to disconnect my Bluetooth device and use the program to connect it). No matter what command I sent, the code returned was **AT+BRSF=767**, which almost depressed me. Is there something wrong? Ask for a solution. Thank you.
[image]
[image]

pqviet07, 2022-06-10:
@peterfoot I asked on Stackoverflow about this problem: https://stackoverflow.com/questions/72557503/create-sco-client-profile-driver-for-call-control-from-bluetooth-headset