inthehand/32feet #196: MAUI IOS Use InTheHand.Net.Bluetooth  Error This Functionality is not implemented in the protable version of this assembly

jason-apj, 2022-06-09:
Hello,
I currently use inthehand Net. There is a problem with Bluetooth. I hope I can use Bluetooth to connect the printer on IOS. When I create a bluetoothclient object in the code, the system throws an exception, "this functionality is not implemented in the protected version of this assembly"。
I don't know what caused this error. I hope you can give me some help.
thank you
[image]
[image]
[image]

jason-apj, 2022-06-09:
[image]

peterfoot, 2022-06-09:
Hi, looking into this it appears the net6.0-ios build is missing from the NuGet. I'll get this fixed and re-released as soon as I can.

jason-apj, 2022-06-10:
> 嗨，看看这个，似乎NuGet中缺少net6.0-ios版本。我会尽快修复并重新发布。

thank you very much。

jason-apj, 2022-06-10:
> Hi, looking into this it appears the net6.0-ios build is missing from the NuGet. I'll get this fixed and re-released as soon as I can.
hi，I'm having some other problems，I want the bluetooth picker in ios to display the device ID instead of the name ，But I don't know how to do that, the Microsoft documentation is a little sparse on this, right 。I have a bunch of bluetooth printers of the same model，If the picker displays the name, obviously I can't find the device I want to connect to 。If possible, could you give me some ideas? 
Thank you very much.

peterfoot, 2022-06-10:
Hi, the problem with this is that iOS doesn't have a classic Bluetooth API and so the implementation uses the ExternalAccessory API. You cannot access the raw Bluetooth address on iOS so the only meaningful identifier is the name. This is the same with their Bluetooth LE implementation where you can get a unique UUID for the device but not read the raw Bluetooth address. Some printers allow you to configure the name (although this could be a pain to reconfigure a large number) so you could include the last few digits of the Bluetooth address or give them some other unique identifier.

jason-apj, 2022-06-10:
> InTheHand.Net.Bluetooth

I'm sorry, because I think my code can be connected to the printer and used these days. After I found inthehand Net. When Bluetooth does not support Maui IOS for the time being, I copied the relevant code from your source code to my project and modified the error reporting code. When I use the picker, I find that it seems that it can only scan the Bluetooth printer device. For other Bluetooth devices, it does not appear in the picker. In fact, I can see many Bluetooth devices in the mobile Bluetooth. I have not modified the code of the picker。
[image]

peterfoot, 2022-06-10:
The picker uses the ExternalAccessory picker. This will only display devices which are Apple MFi compatible - many Bluetooth Printers are (some requiring a change of mode) but other Bluetooth devices will not display. If your Bluetooth classic device doesn't support MFi (And hence the ExternalAccessory API) it cannot be used by 32feet.

jason-apj, 2022-06-10:
> The picker uses the ExternalAccessory picker. This will only display devices which are Apple MFi compatible - many Bluetooth Printers are (some requiring a change of mode) but other Bluetooth [...]

Thank you very much for your answer, which has helped me a lot.。
I noticed that you have a two second waiting time after the Bluetooth selector is selected. I tried to reduce this time to 0.2 seconds. Then when I directly clicked cancel instead of selecting any Bluetooth device, an error occurred. I am currently trying to reproduce this problem normally。
[image]

jason-apj, 2022-06-27:
MAUI ios now works with zebra Bluetooth printers

yylyhldev, 2022-06-30:
> MAUI ios now works with zebra Bluetooth printers
Can you provide a demo of MAUI-Bluetooth

jason-apj, 2022-06-30:
> > MAUI ios 现在可与斑马蓝牙打印机配合使用，能否提供 MAUI 蓝牙演示

> `        #region Connect
               private EAAccessory eAAccessory = null;
    private EASession _session = null;
    EAAccessoryManager.SharedAccessoryManager.RegisterForLocalNotifications();
            try
            {
                await EAAccessoryManager.SharedAccessoryManager.ShowBluetoothAccessoryPickerAsync(null);
}
            catch (NSErrorException ex)
            {
                
            }
            finally
{
    EAAccessoryManager.SharedAccessoryManager.UnregisterForLocalNotifications();
}
if (EAAccessoryManager.SharedAccessoryManager.ConnectedAccessories.Length > 0)
{
    foreach (var accessory in EAAccessoryManager.SharedAccessoryManager.ConnectedAccessories)
    {
        if (accessory.ProtocolStrings.Contains("Protocol for connecting devices"))
        {
            eAAccessory = EAAccessoryManager.SharedAccessoryManager.ConnectedAccessories[0];
            break;
        }
    }
}

if (eAAccessory != null)
{
    if (_session == null)
    {
        _session = new EASession(eAAccessory, eAAccessory.ProtocolStrings[0]);
    }
    _session.InputStream.Schedule(NSRunLoop.Current, NSRunLoopMode.Default);
    _session.InputStream.Open();
    _session.OutputStream.Schedule(NSRunLoop.Current, NSRunLoopMode.Default);
    _session.OutputStream.Open();

}
#endregion

#region Close
_session.InputStream.Close();
_session.InputStream.Unschedule(NSRunLoop.Current, NSRunLoopMode.Default);
_session.InputStream.Delegate = null;
_session.InputStream.Dispose();
_session.OutputStream.Close();
_session.OutputStream.Unschedule(NSRunLoop.Current, NSRunLoopMode.Default);
_session.OutputStream.Delegate = null;
_session.OutputStream.Dispose();
eAAccessory.Dispose();
_session.Dispose();
_session = null;
	#endregion`

jason-apj, 2022-06-30:
> > MAUI ios 现在可与斑马蓝牙打印机配合使用，能否提供 MAUI 蓝牙演示

Hope you can help @yylyhldev

jason-apj, 2022-06-30:
> > MAUI ios now works with zebra Bluetooth printers
> > Can you provide a demo of MAUI-Bluetooth

@yylyhldev

closed 2022-06-30 by jason-apj

yylyhldev, 2022-07-01:
> > > MAUI ios now works with zebra Bluetooth printers
> > > Can you provide a demo of MAUI-Bluetooth
> 
> Android or ios?

both