inthehand/32feet #198: Does not now support interactive use in (Windows) Powershell?

wisemoth, 2022-06-22:
I used to use an old version of this interactively in Powershell.  It is extremely useful for interactive exploration of remote devices.

However, using 4.0.30:
```
> [InTheHand.Net.Sockets.BluetoothClient]::new()
Exception calling ".ctor" with "0" argument(s): "This functionality is not implemented in the portable version of this
assembly. You should reference the NuGet package from your main application project in order to reference the
platform-specific implementation."
At line:1 char:1
```

Which seems to have come from a base implementation at 

https://github.com/inthehand/32feet/blob/73a3fc215f23e633d7c0f8b9d697233f18cb2b22/InTheHand.Net.Bluetooth/Platforms/Standard/BluetoothClient.standard.cs#L18

It would be wonderful if I could use the assembly in Powershell.

peterfoot, 2022-06-22:
The big change from v3 to v4 was instead of a single .NET Framework dll the NuGet package contains runtime versions for multiple platforms. The .NET 4.x runtime dll should support Powershell in the same way. How are you extracting the dll from the NuGet package to use in Powershell?

wisemoth, 2022-06-22:
Thanks @peterfoot .  Your question actually just prompted my to try the `net461` dll in the `nupkg`, and that works.  The problem I mentioned was when I had imported the `netstandard2.0` dll.

In each case I just extract the dll(s) (one in the case of this package) to some location on disk, the `import-module` it (your package wonderfully has no deps to worry about).  In this case it works for `net461` but not `netstandard2.0`.  I have successfully used this approach before with other packages, and usually go for the `netstandard` versions just to assist any forward migration I might be doing.

peterfoot, 2022-09-11:
The netstandard2.0 dll contains no implementation and just allows you to program against the API with the correct platform version being dropped in at build time. I'll see if there is an easier way of facilitating this in the future.

closed 2022-09-11 by peterfoot