inthehand/32feet #199: 32Feet.net Compatibility with Xamarin Android Watch OS

Guangyu-Joshua-Feng, 2022-06-22:
When I install 32feet to Xamarian for Android, the follow errors occur:

Error	NU1202	Package 32feet.NET 3.5.0 is not compatible with monoandroid12.0 (MonoAndroid,Version=v12.0). Package 32feet.NET 3.5.0 supports:
  - net (.NETFramework,Version=v0.0)
  - net-cf (.NETFramework,Version=v0.0,Profile=CompactFramework)	

How should I resolve this?

peterfoot, 2022-06-22:
The package InTheHand.Net.Bluetooth is the current version and supports Android

Guangyu-Joshua-Feng, 2022-06-24:
Works out great! Thank you Peter.

Guangyu-Joshua-Feng, 2022-06-24:
I used the code that works completely compatible fine with 32feet.net and run it with the libaray you mentioned above. At the following lines I got an error:
```
BluetoothEndPoint ep = new BluetoothEndPoint(blueAddress, _serviceClassId);

Console.WriteLine("Connecting!");
client.Connect(ep);
```
**Error Message**:
Error	CS7036	There is no argument given that corresponds to the required formal parameter 'service' of 'BluetoothClient.Connect(BluetoothAddress, Guid)'	App2	C:\Users\DBS\Desktop\DBS\First_Xmarin_Android_Watch_App\App2\MainActivity.cs	76	Active

So is there a way to get around this error caused by the package change?

peterfoot, 2022-06-24:
client.Connect(blueAddress, _serviceClassId);

Guangyu-Joshua-Feng, 2022-06-25:
Thank you so much for your help Peter! The problem above is resolved in the way you suggested. 

Meanwhile, I ran into another issue caused by the package change that I have no clue after reading your code samples:

At the following lines I got an error: Visusal Studio highlighted "client.DiscoverDevices()" when error occurs at the debug mode.
```
Bluetooth client = new BluetoothClient();
for each (var item in client.DiscoverDevices()) {
    ......
}
```
**Error Message:**
System.NullReferenceException: 'Object reference not set to an instance of an object.'

So is there a way to get around this error?

Guangyu-Joshua-Feng, 2022-06-27:
My code is the following and wanna run it on watch. The line  client.DiscoverDevices() fails:
```
#define isServer
public class MainActivity : WearableActivity
    {
        TextView textView;
        int count = 0;
        protected override void OnCreate(Bundle bundle)
        {
            base.OnCreate(bundle);
            SetContentView(Resource.Layout.activity_main);
            Button button = FindViewById<Button>(Resource.Id.dialog_button);
            textView = FindViewById<TextView>(Resource.Id.text);
            button.Click += delegate {
                
                String sent_message = bluetooth_Send(); // here is where I call the bluetooth sending code.
                textView.Text = string.Format("Message: {}. Sent {} times", sent_message, count++);

            };
            SetAmbientEnabled();
        }

        private string bluetooth_Send()
        {
            String sentMessage = "null";
            Guid _serviceClassId = new Guid("D8D0378635F844FC818FC62C17F63205");

            BluetoothClient client = new BluetoothClient();
            //BluetoothRadio radio = BluetoothRadio.PrimaryRadio;
            //radio.Mode = RadioMode.Connectable;
            BluetoothAddress blueAddress = null;

            IReadOnlyCollection<BluetoothDeviceInfo> devices = null;
            if (client != null)
            {
                 devices = client.DiscoverDevices();
            }
            

            //get devices
            Console.WriteLine("Enumerating Devices until the desired one");
            for (int i = 0; i < devices.Count; i++)
            //foreach(var item in client.DiscoverDevices())
            {
                ...

            }

            return sentMessage;
        }

```

peterfoot, 2022-06-30:
I'm not familiar with Wear OS, but for Android in general have you added the required permissions to your appmanifest and requested the required runtime permission? Some info here - https://github.com/inthehand/32feet/wiki/Permissions

closed 2025-12-14 by peterfoot