inthehand/32feet #202: Basic MAUI sample

TwoTicketsPls, 2022-06-30:
Does anyone know of a basic MAUI sample using `InTheHand.Net.Bluetooth`. Or is this not supported?

With a basic proof-of-concept (scan for devices) I immediately hit a dead end (code and error below). 

There is some indication in the commits/issues that MAUI has been looked at. Still too early, or am I doing something really dumb?

```
   at System.Runtime.InteropServices.Marshal.ThrowExceptionForHR(Int32 errorCode)
   at WinRT.DelegateExtensions.DynamicInvokeAbi(Delegate del, Object[] invoke_params)
   at ABI.Windows.Foundation.IAsyncOperationMethods`1.GetResults(IObjectReference _genericObj)
   at ABI.Windows.Foundation.IAsyncOperation`1.global::Windows.Foundation.IAsyncOperation<TResult>.GetResults()
   at InTheHand.Net.Sockets.BluetoothClient.PlatformDiscoverDevices(Int32 maxDevices)
   at InTheHand.Net.Sockets.BluetoothClient.DiscoverDevices(Int32 maxDevices)
   at InTheHand.Net.Sockets.BluetoothClient.DiscoverDevices()
   at MauiBlutoothTester.MainPage.OnCounterClicked(Object sender, EventArgs e) 
in E:\MauiBlutoothTester\MainPage.xaml.cs:line 27
```

[image]

peterfoot, 2022-06-30:
What platform is this on? Window? Is Bluetooth turned on when you run this and correctly running discovery from Settings?

TwoTicketsPls, 2022-06-30:
Hi @peterfoot ,

I have made a small repo illustrating what I am seeing at https://github.com/TwoTicketsPls/Maui-32-Feet-Issue

It contains bare bones MAUI and WPF versions of the same `DiscoverDevices` call. The WPF one works fine. The MAUI fails.

Windows 11. Latest version of VS2022 Preview. I have only tested Windows as the target platform. Bluetooth is turned on. Running discovery from Settings enumerates the target devices successfully (as does the WPF version).

chno299, 2022-07-23:
Hi TwoTicketsPls,
I Tested yout repo: You have to handle Permissions first!

This is what i get:

[0:] Error searching for devices:	Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@a287fb6f: AdapterService getScanMode

Dont ask me for a Solution i am fighting the same problem

TwoTicketsPls, 2022-08-09:
Thanks @chno299 .

I never got as far as testing on Android, as it bombs for me on Windows with the debug output. 
```
Exception thrown: 'System.Runtime.InteropServices.COMException' in Microsoft.Windows.SDK.NET.dll
Error searching for devices:	A method was called at an unexpected time.
```

Stack Trace:
```
   at WinRT.DelegateExtensions.DynamicInvokeAbi(Delegate del, Object[] invoke_params)
   at ABI.Windows.Foundation.IAsyncOperationMethods`1.GetResults(IObjectReference _genericObj)
   at InTheHand.Net.Sockets.BluetoothClient.PlatformDiscoverDevices(Int32 maxDevices)
   at Maui_32_Feet_Issue.MainPage.OnCounterClicked(Object sender, EventArgs e) in E:\Maui-32-Feet_Issue\Maui-32-Feet_Issue\MainPage.xaml.cs:line 27
```

TheSkoo, 2022-11-26:
Using .NET MAUI running win forms app I get the same problem
Stack Trace
   at System.Runtime.InteropServices.Marshal.ThrowExceptionForHR(Int32 errorCode)
   at WinRT.DelegateExtensions.DynamicInvokeAbi(Delegate del, Object[] invoke_params)
   at ABI.Windows.Foundation.IAsyncOperationMethods`1.GetResults(IObjectReference _genericObj)
   at ABI.Windows.Foundation.IAsyncOperation`1.global::Windows.Foundation.IAsyncOperation<TResult>.GetResults()
   at InTheHand.Net.Sockets.BluetoothClient.PlatformDiscoverDevices(Int32 maxDevices)
   at InTheHand.Net.Sockets.BluetoothClient.DiscoverDevices(Int32 maxDevices)
   at BuildingSwayLib.Toothy..ctor() in C:\Software\MAUI\BuildingSway\BuildingSwayLib\Toothy.cs:line 16
HResult    -2147483634
A method was called at an unexpected time.

igiona, 2022-11-30:
I get the very same error on Android Emulator, although I think I added all necessary permissions in the Android manifest.

Does anyone has a working example?
It's kind of a killer to get such problems on a sample app provided in the library repo...

peterfoot, 2022-11-30:
There are now two implementations of the windows dll. WPF will by default pick the Win32 version, Maui uses WinUI and will target Windows 10 so will pick the WinRT implementation. There was an issue with how the library calls async methods in a currently sync only API. I've now fixed this for 4.0.32 to get past this block. However there is still a known issue with the WinRT api that it has quite a long timeout when doing a discovery. My plan is eventually to offer a better async alternative using the DeviceWatcher as this will allow you to track devices going in and out of range as well as limit the time taken or number of devices to return (the current implementation takes a set time even if you request only 1-2 devices).

The Android issue is unrelated - for recent versions of Android you have to explicitly request permission from the user (as well as having the correct permissions in the manifest). I have some code for this which plugs into the MAUI permissions system which I will be releasing as a separate NuGet.

closed 2022-11-30 by peterfoot

igiona, 2022-11-30:
@peterfoot do you have a reference to the Android-permission system code?
If not, it would be great if you could keep this thread posted when your android-MAUI-permissions package gets released 👍 

Thank you for your hard work!