inthehand/32feet #210: Way to bypass Windows 10 pair device permission?

hasham7861, 2022-08-07:
Context: I am trying to make an application which can quickly connect to a bluetooth device. As soon as I make a connect request. I get this Pair a device notification before allowing me to connect. 

Question: I am wondering if I can auto accept this permission for connecting the device on windows.

[image]

peterfoot, 2022-08-09:
BluetoothSecurity.PairRequest can automate the pairing process with a known PIN or numerical value.

hasham7861, 2022-08-11:
Thank you @peterfoot one question though in the windows bluetooth settings, it says my headphones are connected but the device itself does not receive any notification that it is connected already. Is there a helper to communicate to signal the device that it is now connected. 

My code is now this:

```
// pair of the device
BluetoothSecurity.PairRequest(device.deviceAddress, inputPin);
// categorize into audio section for windows
device.setServiceState(BluetoothService.Handsfree,true);
```