inthehand/32feet #211: [regression] NullReferenceException in Bluetooth.ScanForDevicesAsync() when no RequestDeviceOptions passed

MadManMarkAu, 2022-08-16:
Created a new project (.NET Framework 4.8, VB.NET)
Added InTHeHand.BluetoothLE NuGet (latest version, 4.0.24)
Called Device.ScanForDevicesAsync() from async button click event.
Crashes with a NullReferenceException.

If I call the same procedure, but pass in a new RequestDeviceOptions class, it works fine. It's only the overload for ScanForDevicesAsync() with no RequestDeviceOptions that crashes.

This is a regression, as it worked in 4.0.23.

Code:
```vbnet
Imports InTheHand.Bluetooth

Public Class Form1
    Private Async Sub butScan_Click(sender As Object, e As EventArgs) Handles butScan.Click
        Await Bluetooth.ScanForDevicesAsync()
    End Sub
End Class
```

Exception:
```
System.NullReferenceException: 'Object reference not set to an instance of an object.'
```

Call stack:
```
    InTheHand.Bluetooth.Bluetooth.PlatformScanForDevices(InTheHand.Bluetooth.RequestDeviceOptions)
    System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(System.Threading.Tasks.Task)
    System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(System.Threading.Tasks.Task)
    System.Runtime.CompilerServices.TaskAwaiter<TResult>.GetResult()
    BeaconConfig.Form1.butScan_Click(Object, System.EventArgs) in Form1.vb
```

peterfoot, 2022-08-16:
I can see where this was introduced in the windows code. Will get it fixed for the next release

monkeyfuckeddog, 2022-08-19:
I got the same problem.

peterfoot, 2022-09-09:
Resolved in 4.0.25

closed 2022-09-09 by peterfoot

monkeyfuckeddog, 2022-09-09:
Got it！