inthehand/32feet #212: When call DiscoverDevices() -- System.Runtime.InteropServices.COMException: 'A method was called at an unexpected time.'

VolodymyrTrishynGL, 2022-08-16:
**OS**

Edition	Windows 10 Pro
Version	21H2
OS build	19044.1889
Experience	Windows Feature Experience Pack 120.2212.4180.0

**Project**
```
<Project Sdk="Microsoft.NET.Sdk">
  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>net6.0-windows10.0.19041.0</TargetFramework>
    <ImplicitUsings>enable</ImplicitUsings>
    <Nullable>enable</Nullable>
  </PropertyGroup>
  <ItemGroup>
    <PackageReference Include="InTheHand.Net.Bluetooth" Version="4.0.30" />
  </ItemGroup>
</Project>
```

**Code**
```
using InTheHand.Net.Sockets;

namespace BluetoothCSharpTest
{
    class Program
    {
        static void Main(string[] args)
        {
            BluetoothClient client = new BluetoothClient();
            var devices = client.DiscoverDevices();
            Console.ReadLine();
        }
    }
}
```

**Result**
[image]

**Details**

```
System.Runtime.InteropServices.COMException
  HResult=0x8000000E
  Message=A method was called at an unexpected time.

  Source=Microsoft.Windows.SDK.NET
  StackTrace:
   at WinRT.DelegateExtensions.DynamicInvokeAbi(Delegate del, Object[] invoke_params)
   at ABI.Windows.Foundation.IAsyncOperationMethods`1.GetResults(IObjectReference _genericObj)
   at InTheHand.Net.Sockets.BluetoothClient.PlatformDiscoverDevices(Int32 maxDevices)
   at BluetoothCSharpTest.Program.Main(String[] args) in D:\Projects\LucasDiscovererEnjine\LucasDiscovererEnjine\Program.cs:line 10
```

VolodymyrTrishynGL, 2022-08-22:
DiscoverDevices() works well with project type Console App (.Net framework).
So I think the problem is UWP applications.

FifthCloud, 2022-11-20:
Have put that code within a Class Library or even directly into the Console App and I am getting the same exact error for "DiscoverDevices()". 

Not exactly sure what to do in this scenario. Seems to me that it might have something to do with sync or async? Guessing here.