inthehand/32feet #214: How can I use this project in unity

monkeyfuckeddog, 2022-08-19:
I am programming with unity. Now I want a bluetooth module. How can I use the library in unity? When I move the InTheHand.BluetoothLE.dll and call the api in unity. I got error blow.

PlatformNotSupportedException: Operation is not supported on this platform.
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () (at <695d1cc93cca45069c528c15c9fdd749>:0)
System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) (at <695d1cc93cca45069c528c15c9fdd749>:0)
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) (at <695d1cc93cca45069c528c15c9fdd749>:0)
System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) (at <695d1cc93cca45069c528c15c9fdd749>:0)
System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () (at <695d1cc93cca45069c528c15c9fdd749>:0)
NewBehaviourScript+<TestDeviceDiscovery>d__1.MoveNext () (at Assets/NewBehaviourScript.cs:17)
Rethrow as AggregateException: One or more errors occurred.
System.Threading.Tasks.Task.ThrowIfExceptional (System.Boolean includeTaskCanceledExceptions) (at <695d1cc93cca45069c528c15c9fdd749>:0)
System.Threading.Tasks.Task.Wait (System.Int32 millisecondsTimeout, System.Threading.CancellationToken cancellationToken) (at <695d1cc93cca45069c528c15c9fdd749>:0)
System.Threading.Tasks.Task.Wait () (at <695d1cc93cca45069c528c15c9fdd749>:0)
NewBehaviourScript.Start () (at Assets/NewBehaviourScript.cs:13)

What should I do when I want use the library in untiy? Or how can I package a .net library that can use in unity.

monkeyfuckeddog, 2022-08-19:
The dll file I was using is under .netstanded2.0.  InTheHand.BluetoothLE.dll

peterfoot, 2022-09-09:
The .netstandard dll has no functionality. You have to use the platform specific dll for this to work. I realise this makes it difficult to use Unity for a multi-platform project but it doesn't resolve the dlls for build in the way that Visual Studio does when you add a NuGet package and packages the correct platform dll at build time.