inthehand/32feet #222: System.ArgumentException: Value does not fall within the expected range.

Karmageddonium, 2022-09-16:
First of all thank u for this gorgeous lib :)

Windows 10 Pro
21H2
25.‎05.‎2021
19044.1889
Windows Feature Experience Pack 120.2212.4180.0

InTheHand.BluetoothLE 4.0.25
.netCore 3.1

A am using HM-10 Bluetooth module with Arduino

**Getting GattChartacteristic:**
```
  var a = await Bluetooth.ScanForDevicesAsync();
  var device = a?.SingleOrDefault(x => x.Name == "BT05");
  Debug.WriteLine("MainPage");
  bool availability = false;
  while (!availability)
  {
      availability = await Bluetooth.GetAvailabilityAsync();
      if(!availability)
      {
          await Task.Delay(500);
      }
  }
  foreach (var d in await Bluetooth.GetPairedDevicesAsync())
  {
      Debug.WriteLine($"{d.Id} {d.Name}");
  }
  Bluetooth.AdvertisementReceived += Bluetooth_AdvertisementReceived;
  if (device != null)
  {
      device.GattServerDisconnected += Device_GattServerDisconnected;
      await device.Gatt.ConnectAsync();

      var servs = await device.Gatt.GetPrimaryServicesAsync();

      var myServ = servs.SingleOrDefault(x => x.Uuid.ToString() == "FFE0");
      outputCharacteristic = (await myServ.GetCharacteristicsAsync()).SingleOrDefault(x => x.Uuid.ToString() == "FFE1");
  }
```
**TestCase:**
```
List<int> erors = new List<int>();
for(int i = 1; i <= 512; i++)
{
    try
    {
        byte[] source = Encoding.UTF8.GetBytes(Enumerable.Repeat('F', i).ToArray());
        await outputCharacteristic.WriteValueWithoutResponseAsync(source);
    }
    catch(Exception ex)
    {
        erors.Add(i);
    }
}
```
The single known restriction i know that length should not be more then 512, but all data with length >20 causes such an exception.

So if i try to send 1-20 bytes at once - all is ok, i see data transfer on tx bus
But 21-512 throws an error

```
"   в System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   в System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   в System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   в System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd(Task task)
   в System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
   в InTheHand.Bluetooth.GattCharacteristic.<PlatformWriteValue>d__39.MoveNext()
   в System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   в System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   в System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   в System.Runtime.CompilerServices.TaskAwaiter.GetResult()
   в TachoArduinoClient.MainWindow.<Button_Click>d__12.MoveNext() в C:\\Users\\Aleksandr\\source\\repos\\TachoArduinoClient\\TachoArduinoClient\\MainWindow.xaml.cs:строка 198"
```

peterfoot, 2022-09-16:
Have you checked the Mtu property on device.Gatt? Unfortunately on the Windows API you can't request a higher Mtu but it should automatically negotiate with the remote device to get the best value.

Karmageddonium, 2022-09-16:
You are right, MTU is set to 23 in Gatt, but it is still more than 20 anyway))) I am a bit confused. Simple negotiation will do not make a deal. If i send 23 bytes it will throw again.  Unless i do not know about some 3 "system info" bytes like headers or a something like this which is hidden in request.

Otherwise, i am a programmer of both sides of my system, so i can send even byte by byte with mtu 1. 
So this exception is not so