inthehand/32feet #224: Bluetooth.AvailabilityChanged doesn't work on Windows

NilsHoyer, 2022-10-06:
I want to get invoked, if the Bluetooth feature of my Windows system changed (by disabling/enabling it manually).

To do this, I'm trying to use the `Bluetooth.AvailabilityChanged` event of the `InTheHand.BluetoothLE` package (4.0.25):

    Bluetooth.AvailabilityChanged += Bluetooth_AvailabilityChanged;

... with this method:

    protected void Bluetooth_AvailabilityChanged(object? sender, EventArgs e)
    {
        ...
    }

Unfortunately, this method will not be executed. Any idea, why?

I'm working with Windows 10 Pro 21H2 19044.2075 on a 10 years old MacBook Pro device.

NilsHoyer, 2022-10-07:
This works on my end (Windows 10 Pro 21H2 19044.2075) with .Net 6.0.9.31619 on 64 bit:

    var deviceInformations = await DeviceInformation.FindAllAsync(BluetoothAdapter.GetDeviceSelector());
    foreach (var deviceInformation in deviceInformations)
    {
        var bluetoothAdapter = await BluetoothAdapter.FromIdAsync(deviceInformation.Id);

        var bluetoothRadio = await bluetoothAdapter.GetRadioAsync();
                bluetoothRadio.StateChanged += BluetoothRadio_StateChanged;
    }

...

    private void BluetoothRadio_StateChanged(Radio sender, object args)
    {
        ...
    }

peterfoot, 2022-10-10:
I've made some updates to the code for this so that it will be in the next NuGet release. For now using the Bluetooth Classic code is a good workaround

closed 2022-10-10 by NilsHoyer