inthehand/32feet #227: Missing Rfcomm device for the serial port in Windows

NilsHoyer, 2022-10-10:
I'm trying to connect from Windows to my Android device via the serial port with this code:

    var localClient = new BluetoothClient();
    foreach (BluetoothDeviceInfo device in localClient.PairedDevices)
    {
        var address = BluetoothAddress.Parse(device.DeviceAddress.ToString());
        var endPoint = new BluetoothEndPoint(address, BluetoothService.SerialPort);
        BluetoothClient remoteClient = new();
        try
        {
            remoteClient.Connect(endPoint);
        }
        catch (SocketException e) when ((uint)e.SocketErrorCode == 10049) // AddressNotAvailable / WSAEADDRNOTAVAIL 
        {
            // This happens!!!
        }

While this, I get the SocketException marked with "// This happens!!!".

But after I executed the "MyPhoneExplorer" app and connect to the same pair, this problem does not exists.

To hunt this problem, I used the following code:

    var btRemoteDevices = await DeviceInformation.FindAllAsync(RfcommDeviceService.GetDeviceSelector(RfcommServiceId.SerialPort));

Before I execute the "MyPhoneExplorer" app, the result of it is empty. But after executing it, it's filled:
[Screenshot](https://i.ibb.co/xjjjGgs/Unbenannt.png)

The name of this new created device is "MyPhoneExplorer". So that app has added this device.

If I now uninstall my Bluetooth driver inside the Windows Device Manager and reinstall it, I have the same problem as before.

Can 32feet create this device too, or can I do this by myself?

NilsHoyer, 2022-10-16:
I found out, it will be created by calling `GetRfcommServicesForIdAsync`:

    var remoteDeviceInformations = await DeviceInformation.FindAllAsync(BluetoothDevice.GetDeviceSelector());
    if (remoteDeviceInformations != null)
        foreach (var remoteDeviceInformation in remoteDeviceInformations)
            try
            {
                var remoteDevice = await BluetoothDevice.FromIdAsync(remoteDeviceInformation.Id);
                if (remoteDevice.BluetoothAddress.Equals(remoteDevice.BluetoothAddress))
                {
                    var rfcommServices = await remoteDevice.GetRfcommServicesForIdAsync(RfcommServiceId.SerialPort, BluetoothCacheMode.Uncached);

peterfoot, 2022-10-16:
I assume you are using v3.5 of the library? Connect(BluetoothEndPoint) is not used anymore and Connect(BluetoothAddress,Guid) achieves the same result.

NilsHoyer, 2022-10-16:
To connect, I do no use 32feet anymore. I just tried to assist you solving that, what you labeled as a "bug".