inthehand/32feet #240: Rfcomm BluetoothClient just supports Android Devices?

Chrisi1993, 2022-11-04:
Hi,

I have established a Rfcomm connection to a Respberry Pi3B with an Android smartphone. This works also so far.
But IOS seems to be not supported. In Visual Studio I get the following message:

Warning CA1416 This call site is reachable for: Android, 21.0 and higher versions. "BluetoothClient" only supported for: Android, 31.0 and higher versions.

Is this type of connection generally not supported on IOS? Or am I using the library incorrectly?

On Apple's developer page I found the following information about the RFCOMM protocol:
https://developer.apple.com/documentation/iobluetooth/iobluetoothrfcommchannel

But it looks like this is only for MacOS and not IOS.

Does anyone have experience with this and can help me?

peterfoot, 2022-11-04:
The Android warning may indicate that you are referencing the wrong platform dll in your project. There is a BluetoothClient for iOS but it is very restricted because it uses the External Accessory framework. Therefore you can only connect to MFi certified devices, not just any RFComm service.

Chrisi1993, 2022-11-04:
Thanks for your fast response.

I am using this includes at the moment for Android:
using InTheHand.Net.Bluetooth;
using InTheHand.Net.Sockets;
using InTheHand.Net;

Can you tell me which one I need for IOS?

And is there any other possibility to connect to a Bluetooth device without pairing?

If not, is it possible to automate the pairing process in software? For example that the user only has to accept it on his phone. Because the user will have no physical access to the Raspberry PI.

peterfoot, 2022-11-06:
You won't be able to connect to a custom device from iOS - only a certified MFi device. Some printers, barcode scanners and others are supported.

closed 2022-11-30 by peterfoot