inthehand/32feet #245: Unable to scan in Mac and ios In MAUI

akycdi, 2022-11-23:
I am using in the hand in an maui it is working in the windows and android but it is unable to scan the devices in the mac and ios platforms. I have added and given all the permissions that are required.

peterfoot, 2022-11-23:
The Apple platforms only support RequestDeviceAsync currently which presents a dialog to the user to choose an available device. Unfortunately, the code doesn't throw an exception but instantly returns an empty collection. I will get this corrected and the documentation updated with more clarity on platform differences.

akycdi, 2022-11-24:
Hi Thanks for the reply can you give a example or point to the example where the scanning and connection and writing happens in ios and mac framework for maui