inthehand/32feet #246: Bluetooth LE Notifications in MAUI

MrCircuit, 2022-11-24:
I am currently facing an issue with Bluetooth LE Notifications in MAUI on a Windows device. The CharacteristicValueChanged event is not firing, although requested before by calling StartNotificationsAsync on the characteristic (which has the Notify/Indicate flag set).

The advertisement of the device is working and I can check the GATT services and characteristics of the device, which tells me that at least Read calls should be working well.

I followed several paths to possible solutions to problems that describe a similar behaviour (especially this one: https://social.msdn.microsoft.com/Forums/en-US/58da3fdb-a0e1-4161-8af3-778b6839f4e1/bluetooth-bluetoothledevicefromidasync-does-not-complete-on-10015063?forum=wdk). But nothing solved the issue.

And last but not least: Basically the same sequence of calls works in a UWP app (https://github.com/NeoSmartpen/UWP-SDK).

Have you already tested the full set of BLE features with MAUI? Any hint, what could be missing?

Thank you & best regards
Uwe

peterfoot, 2022-12-02:
That forum thread seems to indicate an old Windows 10 bug around BluetoothLeDevice.FromIdAsync so I don't see how it is related. I assume you have the Characteristic and you can read the value successfully. Does the StartNotifications() call complete without errors or anything odd in the debug window? Are you calling it from the UI thread or a worker thread?

MrCircuit, 2022-12-11:
StartNotifications completes without errors. I called it from from a "Connect" button click event but after several async/await steps, like this strongly simplified code shows:

```
            var gatt = selected.Gatt;
            Debug.WriteLine($"Device: {selected.Name}");

            if (!gatt.IsConnected)
                await gatt.ConnectAsync();

            var services = await gatt.GetPrimaryServicesAsync();

            foreach (var service in services)
            {
                Debug.WriteLine($"[{service.Uuid}] Service");
                Characteristics = (await service.GetCharacteristicsAsync()).ToList();

                foreach (var characteristic in Characteristics)
                {
                    Debug.WriteLine($" - [{characteristic.Uuid}] Characteristic - {characteristic.Properties}");

                    if (characteristic.Properties.HasFlag(GattCharacteristicProperties.Notify))
                    {
                        await characteristic.StartNotificationsAsync();
                        characteristic.CharacteristicValueChanged += OnValueChanged;
                    }
                }
            }
```

The target device has just one service with one nofify property. The OnValueChanged handler never gets triggered.

Do you think, that this behaviour may result from calling StartNotificationsAsync from the wrong working thread?

An now - to make things even more complicated: To double check the behaviour, I took another completely different BLE device and checked again. In this case it worked! So the question is - why does the target device (this NeoSmartPen BLE thingy I mentioned) only work with a UWP app? Is there any difference in the handling that I may have overlooked?

peterfoot, 2022-12-12:
Does the characteristic properties contain Notify or Indicate? Currently the StartNotifications method doesn't throw an error if one of these is not present and will just return. Arguably this should be changed to rule out that as a cause of not receiving events. I'm not sure why it would work with one device and not another. Is it paired with another device which is already using notifications?

MrCircuit, 2022-12-21:
After some deeper research on the situation I found out what makes the difference between the device that works and the digital pen that does not: The former sends notifications spontaneously (or more precisely by user interaction) while the latter sends notifications not before receiving a magic packet.

In my more comprehensive test, this magic packet was sent and it failed too. What I learned from this was to question the send process rather than the notification itself. Therefore I took the UWP code I mentioned before and followed the routines to the final send step by step in relation to your code and I tracked it down to the following difference:

UWP:
`characteristic.WriteValueAsync(data.AsBuffer());`

InTheHand (windows platform code, with requireResponse set to false):
`_characteristic.WriteValueAsync(value.AsBuffer(), requireResponse ? Uap.GattWriteOption.WriteWithResponse : Uap.GattWriteOption.WriteWithoutResponse);`

I made a cross check and changed the UWP code to yours and now it failed there also!

The question now is: Is this a general issue of the WriteValueAsync function with the GattWriteOption parameter or is it a specific issue of this particular digital pen device?

peterfoot, 2022-12-21:
Are you sure the characteristic supports WriteWithoutResponse? You can check the Properties property for the supported operations. Try switching to WriteValueWithResponseAsync instead.

MrCircuit, 2023-01-03:
Sorry for the late answer. Thank you very much for this important hint. I really forgot to check on that.
Happy New Year!

closed 2023-01-03 by MrCircuit