inthehand/32feet #247: Is there a way to change the MITM protection without altering the source code?

brofloyd, 2022-12-01:
It looks like the authentication setting can only be altered by editing the BluetoothSecurity.win32.cs file. Is there another way around this other than altering the source code?

peterfoot, 2022-12-02:
Given the behaviour on other platforms I think the solution is to change the default to MITMProtectionRequired for all calls to PairRequest. I'll make sure this is implemented in the next release.

brofloyd, 2022-12-02:
Thank you for the response Peter. Would it be possible to expose this setting so that it can be accessed via the BluetoothSecurity class? Possibly as an overload for PairRequest? Example: PairRequest(BluetoothAddress device, string pin, **BluetoothAuthenticationRequirements.XXXXX**).
I think this would be beneficial because there are different levels of security that are used by different devices.

brofloyd, 2022-12-02:
If you would like, I can put together a PR that would accomplish this.

peterfoot, 2022-12-02:
This would be interesting although I don't think there is an equivalent on all platforms so it would need to be implemented so the parameter is ignored on platforms where you can't specify.

brofloyd, 2022-12-02:
Sounds good to me. I'll write something up this weekend. Feel free to include it into the source code if you like it.

peterfoot, 2022-12-02:
That would be really good thanks!

brofloyd, 2022-12-05:
Hey Peter. Just wanted to give you an update. I've been working on the code but I don't have the PR ready just yet. Hopefully I can  have done by tomorrow.

brofloyd, 2022-12-05:
The PR has been submitted. If you would like for me to change anything about it, just let me know!

peterfoot, 2022-12-06:
Thanks, I will take a look