inthehand/32feet #257: Using InTheHand.BluetoothLE.dll in .NET6

kobelev-dev, 2023-01-05:
Dear sirs, I created a simple .Net6 project and added your Nu-Get library there. I am getting:
[image]

Is there any way to do something around this?
Should I maybe try to use the WinRT directly?

Thanks,
Kirill

peterfoot, 2023-01-05:
What is your target platform for the project. Because the implementation is platform specific the NuGet contains net6.0-android, net6.0-ios, net6.0-windows10.0.19041.0 etc. If you try to target an older version of Windows or generic .net6.0 there is no implementation and so it will step down to an earlier .net version dll. While technically Bluetooth LE is supported in older Windows 10 SDKs there are some differences and so that version was selected as a baseline.

kobelev-dev, 2023-01-05:
Thanks for your prompt response. I found what was wrong. I changed from this

[image]

 to this:
 
[image]

After that is started to compile.
By default Visual Studio is not setting the target OS. IMO this piece should be mentioned somewhere in the 32feet docs or Wikis.

closed 2023-01-07 by kobelev-dev

peterfoot, 2023-12-05:
@nref because the library gets its WinRT bindings from having that runtime type set on the assembly it's exposed with the same TFM. The BluetoothLE apis are only available on Windows 10, and the assembly is already marked as compatible down to 10.0.17763.

nref, 2023-12-05:
I can see now that I am [preaching to the choir](https://inthehand.com/2023/08/21/reinventing-the-wheel-again-bluetooth-on-linux/).  🙂