inthehand/32feet #258: ESP32 with 32feet on Windows 11, Bluetooth classic, can't connect

schmitz-u, 2023-01-07:
### Background: 
Finally, I want to use 32feet for connecting an ESP32 device (BT Classic) with Unity 3D (somewhere I found, that someone did this successfully, but no details provided). Since this did not work right way, as a simplified initial step, I used the 

- 32feet (`id="InTheHand.Net.Bluetooth" version="4.0.33.1"`) example “BluetoothClassicConsole” and the 
- ESP32 Example “SerialToSerialBT” 

to get it up and running.
The ESP32 has been paired under Windows system settings -> OK
Using the Windows application “Bluetooth Serial Terminal”, I can connect to the ESP32 and can exchange data in both directions ->OK
### Error:
**Under 32feet I can’t connect to the ESP32**. Error message: 

> System.Net.Sockets.SocketException,   HResult=0x80004005
> Message=A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond 4022D8EAB23E:00001101-0000-1000-8000-00805f9b34fb (0)
> 

###  Details:
The 32feet function `“DiscoverDevices()”` properly discovers the ESP32. The correct device name and address is retrieved.
The following: `client.Connect(device.DeviceAddress, BluetoothService.SerialPort)` fails with the exception mentioned above.

Before going in more depth, I like to clarify the system requirements. The32feet documentation (https://github.com/inthehand/32feet) clearly says: Bluetooth support requires .NET Framework v4.6.1 for desktop Windows 7, 8 and **10**. 
So, **no Windows 11**, correct? In that case we can stop for now. Any plans to support W11 in the future?

I found another thread  #202 , which is a different issue, but the guy reported, that he is also working under **Windows 11**.

Another “violation” to your spec is, that my system has .NET Framework v4.7.1 installed. When trying to install 4.6.1 I get the message “.NET Framework 4.6.1 or a later update is already installed on this computer.”. So I assume using 4.7.1 is Ok.

Would be great to get that thing clarified/fixed.
Thanks a lot for providing all these efforts in the maintenance of 32feet and providing it to public!!

peterfoot, 2023-01-07:
The requirements text needs clarifying. For .NET Framework the minimum supported version is 4.6.1 but you can use the dll against any subsequent version - no need to install an older version.
There are two underlying implementations for Windows:-
Win32 supported on Windows 7 and above
WinRT supported on Windows 10 and above

.NET Framework 4.x will always have the Win32 implementation. 
.NET 5.0 and above and targeting Windows 10.0.19041 or higher use the WinRT implementation. Older targets than this fallback to the .NET 5.0 Windows 7 Win32 version.
The UWP dll uses WinRT.
There are currently both .NET 5.0 and .NET 6.0 builds but the 6.0 dll doesn't offer any additional features. .NET 7.0 is supported but will use the .NET 6.0 dll.
It's not recommended using the library on .NET Core 3.x but it may work (on non-server Windows devices only).

I know that's a little overcomplicated but hopefully it explains it. In short you can use it with .NET 4.7.1 with no problem.

schmitz-u, 2023-01-07:
Thanks for the super-fast response. So seems my general setup is ok and I should stay with .NET (Framework) 4.7.1, since my final target system is Unity 3D and it (currently) only supports .NET Framework - not .NET (Core). 
I will do some deeper investigations now, but since I am not a BT expert to be honest I do not know where to start ;-(. As I said:
- works with a Windows application “Bluetooth Serial Terminal” => so I conclude, the general Windows setup (driver, dongle, ...) and ESP32 side is basically ok
- does not work with 32feet (“DiscoverDevices()” works, but timeout during connect)

Did you observe s.th. comparable before? Any hint in which direction to investigate?

I will study your documentation and check, whether I find. s.th. there. 
Thanks again for your support.

jfellien, 2023-01-08:
@schmitz-u that's great to see you are playing the same setup as I do. In my case I was able to connect Unity3d with the ESP32. On the ESP I've used the sample: https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/SerialToSerialBT/SerialToSerialBT.ino
and in my Unity environment I've entered in the Start method some code of 32feet sample: https://github.com/inthehand/32feet/blob/main/Samples/BluetoothClassicConsole/Program.cs

DiscoverDevices works well and detects the ESP32. After that I store the BluetoothDeviceInfo of my ESP32 to create a connection via BluetoothClient instance. You should not use BluetoothSecurity.PairRequest with a PIN, the communication will not work after that. Its because you can't confirm the pin and Windows is using a different PIN.

BUT, after all I was able to start Unity only one time. The second start blocks the editor, so I have to kill the Unity editor. Why is it so? I found out that the BluetoothClient close and any other disconnect methods will not keep free the used port. It seems the Unity Editor claims the port after closing the "Game".

I'm still in investigation how to solve this. But back to your point. Your setup should work, my is exactly the same (Win 11 and so on) and it works.

jfellien, 2023-01-08:
@schmitz-u btw, I found an example for your idea: https://github.com/jcs090218/32feet.Unity.Example

schmitz-u, 2023-01-09:
@jfellien, it sounds great, that you made the general connection work.
What worries me is that I think I used the same example code, that you mentioned (SerialToSerialBT.ino, BluetoothClassicConsole/Program.cs).... Have to go into more depth again. Won't be able to do this before the next weekend. In case I will take the initial hurdle, we can stay in contact regarding the Unity problem you mentioned.

> btw, I found an example for your idea: https://github.com/jcs090218/32feet.Unity.Example

I see. Basically the same steps as in BluetoothClassicConsole/Program.cs The key point is: `cli.Connect(mDevice.DeviceAddress, BluetoothService.SerialPort);`
Equivalent to `client.Connect(device.DeviceAddress, BluetoothService.SerialPort);`  in BluetoothClassicConsole/Program.cs.
I confirmed, I read out the correct DeviceAddress, but this `client.Connect` times out....
Will get back on the weekend.

peterfoot, 2023-01-10:
One of the complications with Unity is that while it is .NET 4.6.1 compatible it's the Mono runtime and the Mono version of System.Net.Sockets.Socket doesn't allow Bluetooth connections - there appears to be some extra checking of the endpoint passed  in and so the Win32 implementation of 32feet has a Win32Socket class for when it is used from Mono which P/Invokes into the WinSock APIs. If there is a problem with the behaviour of the Close() method which is causing the socket to be blocked from being used more than once I'll need to investigate this. However if the connect is not working in the first instance I'm less clear on why this might be. 
It might be worth checking that there is nothing else paired with the device which may also try to open a connect to the same service. However that being said, I would expect another error message than a timeout. It might be useful if you could build with the source and step into the code and see where it is getting stuck and if there is any other diagnostic information.

jfellien, 2023-01-10:
@peterfoot you are a great developer. +100

I tested a lot of different things now and found out exactly what you describe. Also a few Async methods don't work well. But in the end you lib is the most stable solution for Unity at the moment. (If the "Game" will run unter Windows)

So, for my project I've solved anything.

jfellien, 2023-01-10:
> @jfellien, it sounds great, that you made the general connection work. What worries me is that I think I used the same example code, that you mentioned (SerialToSerialBT.ino, BluetoothClassicConsole/Program.cs).... Have to go into more depth again. Won't be able to do this before the next weekend. In case I will take the initial hurdle, we can stay in contact regarding the Unity problem you mentioned.
> 
> > btw, I found an example for your idea: https://github.com/jcs090218/32feet.Unity.Example
> 
> I see. Basically the same steps as in BluetoothClassicConsole/Program.cs The key point is: `cli.Connect(mDevice.DeviceAddress, BluetoothService.SerialPort);` Equivalent to `client.Connect(device.DeviceAddress, BluetoothService.SerialPort);` in BluetoothClassicConsole/Program.cs. I confirmed, I read out the correct DeviceAddress, but this `client.Connect` times out.... Will get back on the weekend.
> 
> BTW: which Arduinio ESP32 board package version do you use? I found a number of reports mentioning several BT issues with different versions. Right now I use 2.0.5. The newest one (2.0.6) raises some compile errors.

@schmitz-u  my ESP Board is an AZ Delivery WROOM-32.

I found out (today) that Bluetooth.cpp and .h in Arduino 2.0.6 has a bug. So I changed the stuff I need in the files but I would suggest to use 2.0.5

peterfoot, 2025-07-26:
If you are able to discover, but not connect, I would suggest calling GetRfcommServicesAsync() on the BluetoothDeviceInfo and ensure the services you are expecting are available on the device. Some services will require pairing to connect too, but that depends on the device.

EddyPCcrash, 2025-07-30:
A stable Bluetooth connection with the ESP32 now works. After a complete update of the Arduino IDE to 2.3.6 everything works but only with the ESP32 package version 1.04. With the current version 3.3.0 no Bluetooth connection is possible!!

I use Visual Studio Forms and would like to receive data permanently via Bluetooth. For this I use the integrated timer function. Unfortunately, there is always an error message System.InvalidOperationException: "The operation is not allowed on non-connected sockets."
I can send data from the PC to the Bluetooth device, no problem. I can also receive data, but not within the timer.

EddyPCcrash, 2025-07-31:
More details about my problem above.
I would like to constantly receive data with Bluetooth Classic.`
I request the data with >private void brn_getValue_Click(object sender, EventArgs e)<.
This works, but only one value is always displayed/received. Therefore I wanted to request the data query via the timer. But nothing changes. 
How can I call a method in a timer so that the data reception takes place permanently?

`private void timer1_Tick(object sender, EventArgs e)
{
    label14.Text = "";
    lB_Status.Text = "";
    label14.Text = Rueckgabewert;
    lB_Status.Text = Rueckgabewert;
}

private void brn_getValue_Click(object sender, EventArgs e)
{
    label14.Text = "";
    Client = tB_Client_masked.Text;
    BluetoothAddress deviceAddress = BluetoothAddress.Parse(Client);
    BluetoothEndPoint endPoint = new BluetoothEndPoint(deviceAddress, BluetoothService.SerialPort);
    BluetoothClient client = new BluetoothClient();
    client.Connect(endPoint);
    NetworkStream stream = client.GetStream();
    byte[] dataBytes = Encoding.UTF8.GetBytes("Data to send");
    stream.Write(dataBytes, 0, dataBytes.Length);
    byte[] receiveBuffer = new byte[1024];
   int bytesRead = stream.Read(receiveBuffer, 0, receiveBuffer.Length);
   string receivedData = Encoding.UTF8.GetString(receiveBuffer, 0, bytesRead);
        label14.Text = receivedData;
        Rueckgabewert = receivedData;
    timer1.Start();
    stream.Close();
    client.Close();
  
}