inthehand/32feet #259: Cannot connect to HumanInterfaceDevice

KennethEvans, 2023-01-13:
I have a Bluetooth controller which is something like a joystick. Almost everything else works, but it will not connect as a HumanInterfaceDevice.   It supports Bluetooth 3.0. This is the exception:
```
Exception: System.Net.Internals.SocketExceptionFactory+ExtendedSocketException (10060): A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond. 207340009662:00001124-0000-1000-8000-00805f9b34fb (0)
   at System.Net.Sockets.Socket.DoConnect(EndPoint endPointSnapshot, SocketAddress socketAddress)
   at System.Net.Sockets.Socket.Connect(EndPoint remoteEP)
   at InTheHand.Net.Sockets.BluetoothClient.Connect(BluetoothEndPoint remoteEP)
   at InTheHand.Net.Sockets.BluetoothClient.PlatformConnect(BluetoothAddress address, Guid service)
   at InTheHand.Net.Sockets.BluetoothClient.Connect(BluetoothAddress address, Guid service)
   at TabmateBluetoothClient.MainForm.connectClient() in C:\Users\evans\Documents\Visual Studio 2022\Projects\Tabmate Bluetooth Client\Tabmate Bluetooth Client\MainForm.cs:line 117
A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond. 207340009662:00001124-0000-1000-8000-00805f9b34fb (0)
```
The HumanInterfaceDevice service (1124) _is_ installed.  This is what I get for the DeviceInfo:
```
Client Info:
Client: InTheHand.Net.Sockets.BluetoothClient
Client Connected: False

Device Info:
DeviceName: TABMATE
Device Address: 20:73:40:00:96:62
Device Connected: True
ClassOfDevice: 548
Installed Services
    00001000-0000-1000-8000-00805f9b34fb
    00001124-0000-1000-8000-00805f9b34fb
    00001200-0000-1000-8000-00805f9b34fb
```
This is the call I am making:
```
client.Connect(device.DeviceAddress, BluetoothService.HumanInterfaceDevice);
```

peterfoot, 2023-01-13:
When you pair a HID device Windows will setup a connection to it and you will not be able to open a HID connection while it is in use. In this case you can intercept the relevant key presses rather than work directly with the controller device. The Windows.Gaming.Input namespace has functionality for this, or you can use XInput/GameInput APIs.

KennethEvans, 2023-01-13:
Thank you for the quick reply. However, I do not understand it. The device should not be in use. It is also not sending out keystrokes.

To give more detail that I didn't do before. This is a device called TABMATE designed for a particular painting program, Clip Studio Paint. I can assume it is sending information about which button was pressed. The action, in the program, that results from such a press is configurable. The action may also be initiated by a key combination, but that seems irrelevant. A given button can do completely different things in different versions of the program (on different computers, for example), and what it does on any one computer can be changed by the user. 

Since the API is not published, it only works with this program that understands it. My intent is to get which button is pressed, and send the keystrokes myself. I know how to do that part. Then it would work with any program, say Photoshop. The actual key combinations that do common things like increasing the brush size vary from program to program. And to repeat, one button on the device can be configured to do completely different things, increase the brush size, close the program, whatever. 

The other program is not running while I am using 32feet. I don't see how this is different from using different audio programs with the same headset, for example. You can only use one at a time, but you can choose which. 

The word connected is somewhat ambiguous in Bluetooth. The device is definitely paired. When the DeviceInfo says it is connected, I assume that means it is on and in range. The problem is how to talk to it.

It is using Bluetooth 3.0 although BLE would be arguably more appropriate for this low bandwidth task. 

My main experience with Bluetooth is on Android with BLE. I have several apps that successfully access heart monitors. My knowledge of Bluetooth in general is significantly short of yours, however. And I may not understand just how a HID device works. 😉

Thank you for your time.

peterfoot, 2023-01-13:
It won't be the app that connects to the device, it'll be Windows itself. To confirm check the Settings > Bluetooth & devices > Devices list and it probably says connected by it. Because it is not a keyboard but a custom device it may not send any capturable key events but that's why I suggested the other APIs to investigate. You can also try Windows.Devices.HumanInterfaceDevice.HidDevice
Use the static FromIdAsync method once you have a device id - you can find this from Windows.Devices.Enumeration.DeviceInformation.FindAllAsync and look for a name match. If FromIdAsync returns non-null then you have a HidDevice and can receive input reports from it.

KennethEvans, 2023-01-13:
I guess this is the key part I didn't understand.

`It won't be the app that connects to the device, it'll be Windows itself.`

Got it now.  Thank you.

KennethEvans, 2023-02-01:
I have explored this issue further using your suggestions. I was unable to get Windows.Gaming.Input to work. It requires UWP or WinUI in any event. I would rather use Winforms. I have got it working in Winforms  using SharpLibHid. The InTheHand stuff works there also. So I have done what I wanted to do.

The difference between the two is, however, confusing. One identifies the device as TABMATE. The other as a Broadcom device with no mention that I can find of the word Tabmate.

Device Explorer also allows identifies it as Broadcom, with no mention of Tabmate. These seem to be using Winuser.h and User32.dll. InTheHand seems to be using bluetoothapis.h and bluetoothapis.dll (plus others).

The ProductID for this device is longer than a ushort (based on its name in Device Explorer) and thus doesn't conform to the USB spec. The Broadcom device, however, has a ProductID that comforms. (Means someone paid USB some  money.)

So I guess my question is where is InTheHand and the bluetoothapis.h API getting the word TABMATE  and why that does not appear in Device Explorer (that I have found anyway)?

This is not an issue. InTheHand is well done and works well in my opinion (apart from the one issue which is probably related to something I am doing.) However you likeky have the knowledge to answer the question. Thanks in advance. Your other comments above were very valuable.