inthehand/32feet #261: Get "A delegate was assigned when not allowed." when using client.PairedDevices

KennethEvans, 2023-01-16:
I am trying to develop a UWP application.  I am very familiar with WinForms but hardly at all with UWP. (The reason is to be able to use the Game Controller stuff. ) I do have a working app, though. It should be relatively clean and with standard "stuff" as I just created it.

It happens in this block:
```
        public void InitializeBluetooth() {
            client = new BluetoothClient();
            device = null;
            IEnumerable<BluetoothDeviceInfo> pairedDevices = client.PairedDevices;
            foreach (BluetoothDeviceInfo bdi in pairedDevices) {
                if (bdi.DeviceName.Contains("TABMATE")) {
                    device = bdi;
                    break;
                }
            }
```
It gives "A delegate was assigned when not allowed." as the first item of the pairedDevices and throws an exception when the foreach is started.
```
System.InvalidOperationException: 'A delegate was assigned when not allowed.
A delegate was assigned when not allowed.
```
This is when running with "Start Debugging" and looking at the watch for pairedDevices.

When running with "Start Without Debugging" it gives a dialog saying "Unable to activate Windows Store app", which is completely weird.
[image]

The same code runs fine in my WinForms version. The same watch for pairedDevices has a list of the devices as expected. This is standard C#. It should not have anything to do with UWP vs. WinForms. C# is the same. I don't find anything relevant when Googling.

peterfoot, 2023-01-22:
It's a bit of an odd one as there isn't a delegate used inside that property getter. I will need to dig into this further...

KennethEvans, 2023-01-22:
There is also an event in the event viewer.
[image]

m1adow, 2023-08-02:
@peterfoot hi! Sorry for bothering you. Do you have any suggestions about this? I have encountered this error as well

DanDotN3t, 2024-03-29:
Hi! Is there any update on this, I'm hitting the same issue.

Jekops, 2024-07-09:
Hi,

I get the `A delegate was assigned when not allowed exception` on a MAUI application on Windows 10 version 10.0.17763.0 and greater.

This happen in the WindowsBluetoothClient class when the PairedDevices method is called multiple times.

The current implementation seems to create more tasks due to the multiple AsTask involved. This lead in a unhandled task. When the method hit again the new task cannot set the completion callback due to the already existing unhandled task.

@peterfoot Following refactoring of PairedDevices method on WindowsBluetoothClient class solved the issue in my test environment.

- Removed multiple calls of AsTask
- Removed ConfigurationAwait because the Result is joining the current task anyway
- Removed Wait because the Result is already waiting

```
    public IEnumerable<BluetoothDevice> PairedDevices
    {
      get
      {
        var t = DeviceInformation.FindAllAsync(BluetoothDevice.GetDeviceSelectorFromPairingState(true));
        var task = t.AsTask();
        var devices = task.Result;

        foreach (var device in devices)
        {
          var td = BluetoothDevice.FromIdAsync(device.Id).AsTask();
          var bluetoothDevice = td.Result;
          yield return bluetoothDevice;
        }
      }
    }
```

peterfoot, 2024-07-10:
@Jekops thanks for taking your time to investigate this and contribute this solution. Would you like to raise it as a pull request or would you prefer me to integrate it?

Jekops, 2024-07-12:
@peterfoot I tried to push my branch with the suggested solution but I get a denied error from git.

Here is the final fix for WindowsBluetoothClient class.
```
  public IEnumerable<BluetoothDeviceInfo> PairedDevices
  {
      get
      {
          var devicesTask = DeviceInformation.FindAllAsync(BluetoothDevice.GetDeviceSelectorFromPairingState(true)).AsTask();
          var devices = devicesTask.Result;

          foreach (var device in devices)
          {
              var bluetoothDeviceTask = BluetoothDevice.FromIdAsync(device.Id).AsTask();
              var bluetoothDevice = bluetoothDeviceTask.Result;

              yield return new BluetoothDeviceInfo(new WindowsBluetoothDeviceInfo(bluetoothDevice));
          }
      }
  }
```

guerillacodester, 2024-08-22:
will this be fixed in the release? i need to use this library.

guerillacodester, 2024-08-24:
Brilliant! Thanks for the rapid response.