inthehand/32feet #262: Bluetooth.RequestDeviceAsync fails in BluetoothBLE in iOS Maui

mdima, 2023-01-19:
If I call the instruction: 

`currentDevice = await Bluetooth.RequestDeviceAsync(options); `

I receive as error: 

**Message:**
 	System.ArgumentException: 'idx'

**Call stack:** 
 	0x10 in Foundation.NSData.get_Item	C#
 	0x93 in InTheHand.Bluetooth.BluetoothUuid.op_Implicit	C#
 	0xCF in InTheHand.Bluetooth.BluetoothAdvertisingEvent..ctor	C#
 	0x21 in InTheHand.Bluetooth.Bluetooth.OnDiscoveredPeripheral	C#
 	0x5 in InTheHand.Bluetooth.Bluetooth.BluetoothDelegate.DiscoveredPeripheral	C#

The package I use is: InTheHand.BluetoothLE, version 4.0.27. 
The project targets net6.0-ios

Is necessary some other details? 

Thanks,
Michele

peterfoot, 2023-01-22:
I'm afraid I haven't been able to debug this yet as my development environment is broken for .NET MAUI. However I was able to verify the crash. The same code in the latest Xamarin dll is working correctly though.I'll post another update once I've been able to step into the code and investigate properly.

mdima, 2023-01-23:
Thanks @peterfoot !

peterfoot, 2023-02-08:
I've made some changes in the source code which should have this up and running again. I've only tested on .NET 6 so far, not .NET 7 and it's not built into a NuGet yet...

mdima, 2023-02-09:
Thank you @peterfoot ! I will update the package as soon as it will be available on NuGet. Also, for me .NET 6 is perfect...

mdima, 2023-03-02:
Ok, I got the problem. 
Happens when the library finds some device with a null name. :S

peterfoot, 2023-03-02:
Okay, I'll check this out because if it's unable to determine a name it should have a backup identifier.

peterfoot, 2023-10-11:
Fixed - picker now displays identifier if name is null

closed 2023-10-11 by peterfoot