inthehand/32feet #265: Bluetooth.RequestDeviceAsync causes an exception

davPet8842, 2023-01-31:
Hello,

I'm using the BluetoothLe libary under windows in a .net application (.net Framework 4.6.1)

When I'm calling the function Bluetooth.RequestDeviceAsync(), there is thrown an exception "The value is not within the expected range".

The StackTrace is showhing the following value:

StackTrace:   at InTheHand.Bluetooth.Bluetooth.IInitializeWithWindow.Initialize(IntPtr hwnd)" & vbCrLf & "   at InTheHand.Bluetooth.Bluetooth.<PlatformRequestDevice>d__23.MoveNext()" 

What can I do to get this function working?

peterfoot, 2023-01-31:
I'm looking into this because I have seen it myself. What UI framework is your app using? WPF or WinForms?

davPet8842, 2023-01-31:
I'm using WinWorms.

AriDeve, 2023-02-09:
Same error throws using WPF, .net Framework 4.6.1

peterfoot, 2023-02-09:
How are you calling RequestDeviceAsync? From the UI thread, or other? Is your application window visible at the point it is called?

AriDeve, 2023-02-09:
This is the code that does it, was trying to find a package to connect to and read data from ble devices.
On a similar note, seems like `Bluetooth.GetPairedDevicesAsync` and `Bluetooth.ScanForDevicesAsync` never return for me when I tried those in the Test function, and I'm not sure what exactly is wrong.

```
public MainWindow()
{
    InitializeComponent();

    var discoveryTask = Test();
    discoveryTask.Wait();
}

private static async Task Test()
{
    var device = await Bluetooth.RequestDeviceAsync(new RequestDeviceOptions {AcceptAllDevices = true});
    Debug.WriteLine(device);
}
```

peterfoot, 2023-02-09:
Okay, I'm looking into ways to improve this, but in order to call the Windows picker dialog we have to pass a window handle and at the point where InitializeComponent returns the window is not currently visible. You can work around this by adding a handler for the Appearing event and then call the code in there (or run it from a button press on the page).

AriDeve, 2023-02-09:
I'll try it out in a second. Would the same problem apply to the other two methods, or is that a different problem?

AriDeve, 2023-02-09:
It did indeed solve the problem, as well as the other two work now, thank you.

peterfoot, 2023-02-09:
In the case of ScanForDevices, on Windows 10 this uses DeviceInformation.FindAllAsync and there is a hard timeout of 20 seconds before it returns results. I'm looking at replacing this implementation with a watcher so you get realtime updates as each device is found and we can set a custom timeout to return after x devices or y seconds.

closed 2023-10-11 by peterfoot