inthehand/32feet #268: Pairing BluetoothLE device on windows?

AriDeve, 2023-02-10:
How does one go about programmatically pairing BLE device on windows? For my use case this is for remembering devices more than anything, so use can choose to pair a device through the app using `Bluetooth.RequestDeviceAsync` and pairing it automatically, and in the background the app occasionally would check/connect devices from `Bluetooth.GetPairedDevicesAsync()`.

The specific device I'm working with can be paired using Windows bluetooth settings page, but even though `Bluetooth.RequestDeviceAsync` mentions pairing in the pop-up, I'm assuming it's not exactly the right type of pairing.

Couldn't find anything in the docs, except `BluetoothSecurity.PairRequest`, which seems to not work with what I'm, though it may just be that I'm understanding it wrong, as it hangs the process:
`BluetoothSecurity.PairRequest(Convert.ToUInt64(device.Id, 16), "0000");`

peterfoot, 2023-03-04:
Yes, currently the Bluetooth LE library doesn't have its own pairing support. The wording in the connection dialog used with RequestDeviceAsync was recently changed as it makes reference to pairing but it is purely doing discovery and not pairing devices. The way I see this being evolved is to allow automatic pairing when the user selects a device so that it is automated based on the user selection, but we may also need a manual method of pairing for devices retrieved via a scan rather than the dialog.

AriDeve, 2023-03-05:
God it was late when I wrote that one out, and reading it again now, I can see the amount of mistakes there.. 😅

Anyways, thanks for the response, explain everything I wanted to know, and I already made a system for remembering devices myself, so it's not a problem! 👍

closed 2023-03-05 by AriDeve