inthehand/32feet #272: Not supported platform - Windows SDK version 10.0.18362.0

alinatr357, 2023-02-20:
I'm building an aplication that works with Bluetooth and BluetoothLE devices so I use both 'InTheHand.BluetoothLE' and 'InTheHand.Net.Bluetooth' NuGet packages.
I have 2 issues that related to Windows SDK version. 
When I compile the project with SDK version 10.0.19041.0 and above, I have an issue with 'BluetoothSecurity.PairRequest'. When I call the API it returns 'false' right away and the device is not paired. So I tried to compile with SDK version 10.0.18362.0 and it worked smooth, the device got paired.
But then I continued to develop BLE connection and ran into the next issue with 'Bluetooth.ScanForDevicesAsync'. With SDK version 10.0.18362.0 I got 'PlatformNotSupportedException' but again It worked great when I compiled the project with SDK version 10.0.19041.0.
So now I have a problem, because I don't know how to proceed from here. I need both BL and BLE but can't work with the same SDK for both of them. 
Is there something I can do about it?

peterfoot, 2023-02-21:
This looks like an omission in the NuGet metadata rather than an icompatibility. I should be able to fix this in the next release. It should be compatible all the way back to 10.0.16299.0

alinatr357, 2023-02-21:
Thank you very much for your reply :-)
Would you kindly let me know when can I expect the next release?
Thank you.

peterfoot, 2023-06-12:
Supported in InTheHand.BluetoothLE 4.0.033 and InTheHand.Net.Bluetooth 4.0.34

closed 2023-06-12 by peterfoot