inthehand/32feet #283: can not pair and connect device

fuhongbing134, 2023-03-17:
When I use 32feet.net, anything works well. But when I use InTheHand.Net.Bluetooth, it seems that I could not pair and connect with my device.

Here is the function:

private async void DeviceBox_MouseDoubleClick(object sender, MouseButtonEventArgs e)
        {
            var item = DeviceBox.SelectedItem;
            
            foreach (BluetoothDeviceInfo device in _devices)
            {
                if (device.DeviceName == item.ToString())
                {
                    _selectedDevice = device;
                    break;
                }
            }

            if (_selectedDevice == null)
            {
                MessageBox.Show("Please select a device to connect to.");
                return;
            }

            // pair
            await Task.Run(() => BluetoothSecurity.PairRequest(_selectedDevice.DeviceAddress, ""));

            // connect
            await Task.Run(() => _bluetoothClient.Connect(_selectedDevice.DeviceAddress, BluetoothService.SerialPort));

            MessageBox.Show("success");
        }

peterfoot, 2023-03-17:
What platform are you running on, I assume Windows? It would be better to capture the result of the PairRequest call before you proceed to connect - I'm not sure why you are running them in separate Tasks?

fuhongbing134, 2023-03-17:
> What platform are you running on, I assume Windows? It would be better to capture the result of the PairRequest call before you proceed to connect - I'm not sure why you are running them in separate Tasks?

My OS is Windows11.