inthehand/32feet #287: .Net MAUI App connects to HC-05 Module

WongLord, 2023-03-21:
Hello,

I'm making a .Net MAUI App that connects to a HC-05 Bluetooth module.

The idea is to send strings from the HC-05 to the app, I cannot find a way to do it with this package.

I connect to the module with the following code:

```
 foreach (var dev in client.DiscoverDevices())
            {
                if (dev.DeviceName != null)
                {
                    if (dev.DeviceName.Contains("Bike"))
                    {

                        device = dev;
                        DeviceList.Add(device);
                        DeviceSelected = device;
                        IsDeviceListEmpty = false;
                        IsScanning = false;
                        break;
                    }
                }
            }

            if (!device.Authenticated)
            {
                BluetoothSecurity.PairRequest(device.DeviceAddress, "0000");
            }
            device.Refresh();

           client.Connect(device.DeviceAddress, BluetoothService.SerialPort);
```

I'm able to send strings to the HC-05 module with this:

```
var stream = client.GetStream();
StreamWriter sw = new(stream, System.Text.Encoding.ASCII);
sw.WriteLine(data);
sw.Close();
```
Can someone enlighten me with an example of receiving strings from the HC-05 module?

Thank you!

peterfoot, 2023-03-22:
Reading from a device uses the same Stream. You can use a StreamReader for this. However to constantly read from a remote device you'll need to create a loop running on a worker thread so that you don't block the UI - then you can use StreamReader.ReadLine() to read each complete line as it is received.

WongLord, 2023-03-22:
> Reading from a device uses the same Stream. You can use a StreamReader for this. However to constantly read from a remote device you'll need to create a loop running on a worker thread so that you [...]

Hello Peter,

Could you provide an example please?

I tried using this code but is not getting anything.

```
static BluetoothListener listener = new BluetoothListener(BluetoothService.SerialPort);
Task.Run(() => ReceiveData());

private void ReceiveData()
    {
        string commandReceived = "";
        listener.Start();
        while (true)
        {
            client = listener.AcceptBluetoothClient();
            Debug.WriteLine($"RemoteMachineName: {client.RemoteMachineName}");
            if (client.GetStream() != null)
            {
                var stream = client.GetStream();
                StreamReader sr = new(stream, System.Text.Encoding.ASCII);
                commandReceived = sr.ReadLine();
                sr.Close();

                Device.InvokeOnMainThreadAsync(() =>
                {
                    Shell.Current.DisplayAlert($"Command Received", $"Command: {commandReceived}", "OK");
                });
            }
            Task.Delay(1000).Wait();
        }
    }
```

WongLord, 2023-03-23:
Hello,

I was able to read data from the HC-05 Module with the following code:

I started a new Thread after connection:
```
client.Connect(device.DeviceAddress, BluetoothService.SerialPort);

Thread tr = new(ReceiveData);
tr.IsBackground = true;
tr.Start();

private void ReceiveData()
    {
        var stream = client.GetStream();
        byte[] receive = new byte[1024];

        while (true)
        {
            Array.Clear(receive, 0, receive.Length);
            var readMessage = "";
            do
            {
                Thread.Sleep(1000);
                stream.Read(receive, 0, receive.Length);
                readMessage += Encoding.ASCII.GetString(receive);
            }
            while (stream.DataAvailable);
        }
    }
```

closed 2023-03-24 by WongLord