inthehand/32feet #288: WPF +DotNet6

wagenheimer, 2023-03-23:
Is WPF with Dot Net 6 already compatible?

Bluetooth.GetAvailabilityAsync() is always returning null for me.

Does it already work or I need to back to Dot Net 4.8?

wagenheimer, 2023-03-23:
I will need to use the project on Windows 7. Will it work, which configuration should I use?

peterfoot, 2023-03-23:
.NET 6.0 is supported but Windows 10 is required - Windows 7 doesn't have the underlying Bluetooth LE support.

wagenheimer, 2023-03-23:
Thanks for the answer! So there is no way to support BluetoothLE on Windows 7?

Do you have any idea why Bluetooth.GetAvailabilityAsync() is always returning false for me?

I tried to download the source code and then create a new Dotnet6 subproject to be able to debug, but then I can't run it."

"InTheHand.BluetoothLE.dll cannot be referenced because it uses built-in support for WinRT, which is no longer supported in .NET 5 and higher.  An updated version of the component supporting .NET 5 is needed."

"InTheHand.BluetoothLE.dll cannot be referenced because it uses built-in support for WinRT, which is no longer supported in .NET 5 and higher. An updated version of the component supporting .NET 5 is needed.

wagenheimer, 2023-03-24:
I managed to make it work with DotNet 6 by changing the TargetOSVersion to 10.0.19041.0. 

However, I'm not sure if using .NET 4.8 also requires a minimum version of 10.0.19041.0 for BluetoothLE to function properly. Can you confirm this?

peterfoot, 2023-03-24:
The dll is built against the 10.0.19041 SDK so you must target that or higher, however it will run on 10.0.17763.0 and later.

closed 2023-03-24 by wagenheimer