inthehand/32feet #293: Support progress for ObexWebRequest

zhuxb711, 2023-04-15:
ObexWebRequest use its internal MemoryStream currently. We could not get progress and don't have an official way to get the progress. Maybe allow the user to set their own Stream instead of MemoryStream and provide an event for progress handling.

Additive-abc, 2023-09-27:
@zhuxb711  are you able to send a file of more than 100 KB using ObexWebRequest of the Nuget package InTheHand.Net.Obex(4.0.4) if so please help in the below code

using (var bluetoothClient = new BluetoothClient())
{
string filePath = @"D:\pdf_export\Patient.pdf";
var fileName = Path.GetFileName(filePath);
var address = bluetoothClient.PairedDevices.ToList()[1].DeviceAddress; //For my testing purpose only
var uri = new Uri("obex://" + address + "/" + fileName);
ObexWebRequest request = new ObexWebRequest(uri);
request.ReadFile(filePath);
var result = request.GetResponse() as ObexWebResponse;
Console.WriteLine($"Result StatusCode: {result.StatusCode}");

Above code is executed on .net 6 environment. Above code always gives status code 'RequestUrlTooLarge' code always even if the file size is 83kb. when i try to send file of 40KB it works fine.

I have created another dummy application with .net framework dll 'InTheHand.Net.pernosal dll and was able to execute above code without any error.

Thanks in advance.

zhuxb711, 2023-09-27:
@Additive-abc Here is the code for my application (WinUI3). FYR.

@peterfoot Obex should be refactored
1. It's outdated and should support for the file stream directly.
2. It should support for cancellation/abort.
3. It should not read all the byte from file because the file might be very large.
4. It should report the progress about the PUT/GET
5. Async support

``csharp
using (BluetoothDevice Device = await BluetoothDevice.FromIdAsync(<Your device id>))
{
    RfcommDeviceServicesResult PushServices = await Device.GetRfcommServicesForIdAsync(RfcommServiceId.ObexObjectPush, BluetoothCacheMode.Uncached);

    if (PushServices.Error == BluetoothError.Success && PushServices.Services.Count > 0)
    {
        ObexWebRequest ObexRequest = new ObexWebRequest(new Uri($"obex://{BitConverter.ToString(BitConverter.GetBytes(Device.BluetoothAddress).Reverse().ToArray(), 2, 6).Replace("-", string.Empty)}/{Uri.EscapeDataString("<Your file name>")}"));

        using (Stream SendStream = ObexRequest.GetRequestStream())
        {
            using (Stream FileStream = new FileStream("<Your file path>"))
            {
                await FileStream.CopyToAsync(SendStream);
            }

            await SendStream.FlushAsync();
        }

        if (await ObexRequest.GetResponseAsync() is ObexWebResponse Response)
        {
            try
            {
                if (Response.StatusCode.HasFlag(ObexStatusCode.OK))
                {
                    // Handle success status
                }
                else
                {
                    // Handle failure status
                }
            }
            finally
            {
                Response.Dispose();
            }
        }
    }
    else
    {
        throw new NotSupportedException();
    }
}
```

Additive-abc, 2023-09-27:
@zhuxb711 **Thank you** very much for your response. However the code is failing for file of size more than 100KB.

peterfoot, 2023-09-27:
@zhuxb711 there was some work to start this and move to the ObexClient - more closely aligned to HttpClient model rather than the old WebRequest model. It's still on the long list of things to do but equally is open to pull requests etc.

darkhz, 2024-09-23:
Hello, in case anyone is still interested , I've put together a bluetooth-only OPP multiple file transfer implementation with transfer events, based on code from ObexWebRequest.cs and Listener related files. It can most definitely be improved upon, but sending/receiving multiple files works (Windows<->Android (14) at least, when I tested it). 

Do note that this does not follow the WebRequest model.

For the Listener implementation, each file is received to a temporary file first, and then after the file receive is complete, the temporary file is moved to the specified save folder.

The code is here:
https://github.com/bluetuith-org/32feet/blob/main/InTheHand.Net.Obex%2FObexSession.cs

In case you want to use it:
For ObexClient (i.e, sending files from Windows to another device)

```c#
var session = new ObexBluetoothClientSession();
session.SessionEvent += (_, s) => {
Console.WriteLine(s);
};

try {
session.StartSession();
} catch (Exception e) {
Console.WriteLine(e);
return;
}

//repeat any amount of times
session.PutFile(filepath)

session.StopSession()
```

For ObexListener (i.e, receiving files from another device to Windows)

```c#
var session = new ObexBluetoothListenerSession();
// Path to store temporary files during transfer
session.TemporaryFolder = folderpath;
// Path to move the temporary file to after transfer
session.SaveFolder = savepath;
session.SessionEvent += (_, s) => {
Console.WriteLine(s);
};

try {
session.StartSession();
} catch (Exception e) {
Console.WriteLine(e);
return;
}
```