inthehand/32feet #296: 'This functionality is not implemented in the portable version of this assembly.

reyortsed, 2023-04-30:
This is probably a really N00b question, but how do I use this package in VS 2022 .NET 6

I get the error

"'This functionality is not implemented in the portable version of this assembly. You should reference the NuGet package from your main application project in order to reference the platform-specific implementation.'"

I have used NuGet packages a lot but never seen this kind of error

Cheers

//Steve

closed 2023-04-30 by reyortsed

sylnsr, 2023-08-03:
Closed without any comments or solutions ?

peterfoot, 2023-08-03:
I don't know why the poster closed it but the error message is correct and if you don't use a supported platform identifier in your project it will default to the .NET Standard DLL which is a placeholder only. You can't currently target generic net6.0 you have to specify either windows, android or iOS in the consuming project.